openwrt/luci #3613: luci-app-dns  Can not choose URL method to get IP

jsalatiel, 2020-02-07:
OpenWrt 19.0.1
ddns-scripts - 2.7.8-12
ddns-scripts_no-ip_com - 2.7.8-12
luci-app-ddns - 2.4.9-7

luci-app-ddns Change IP address source to URL on advanced settings will never apply/save. 
Steps to reproduce

    Services > Dynamic DNS
    Choose any instance
    Open Advanced Settings
    Field Ip address source [IPv4] = URL
    Press Save & Apply

[image]
The process will timeout and never save.
In log i can see:
`Fri Feb  7 19:08:58 2020 daemon.info uwsgi: invalid CGI response !!!`

jow-, 2020-02-07:
Please retry with uhttpd instead of nginx/uwsgi.

jow-, 2020-02-23:
Timeout.

closed 2020-02-23 by jow-

jsalatiel, 2020-03-01:
Sorry  @jow-  i missed the reply notifications. Yes, it works with uhttpd. is this a nginx/uxsgi bug? can it be fixed ? Thanks!

regulskimichal, 2021-02-23:
Does anyone know how to fix this issue? Alternative DDNS configuration?