openwrt/luci #3616: luci-theme-bootstrap: small visual errors

castillofrancodamian, 2020-02-09:
**There's no space:**

[image]

**And a lot of white space:**

[image]

[image]

[image]

[image]

[image]

[image]

castillofrancodamian, 2020-02-19:
In addition, `Applications` is more "attractive" than `Services`. It is "less" technical.

adelton, 2025-01-10:
I assume the "no space" issue is caused by the notifications on the right getting long and thus them being pushed on a second line, overlapping the page content.

On 23.05.5, when I make the window smaller for the menu and the notifications to fit on one line, I see the "Log out" to jump on the next line and the taller black menu not covering the page content:
[image]

Not sure if this was a deliberate change to address exactly the issue that you reported or a side-effect of some other changes. Can you confirm that you see this problem addressed on latest OpenWrt as well?

adelton, 2025-01-10:
As for "a lot of white space" -- it might not be everyone's cup of tea (I prefer the original OpenWrt theme and this is one of the reasons) but I don't really find anything actionable in the screenshots that you showed, that would scream "the white space here really should be smaller". Can you elaborate if there is something specific you see in there that could be considered a bug?

closed 2025-02-26 by castillofrancodamian