openwrt/luci #3627: L.Requests.request doesn't allow query params with POST

karlp, 2020-02-13:
https://github.com/openwrt/luci/blob/master/modules/luci-base/htdocs/luci-static/resources/luci.js#L721-L733  only allows setting url params for get/options/head, but not for post.  I have an API endpoint that I need to use that needs both url parameters and message body: https://support.dexma.com/hc/en-gb/articles/360007965713-#h_dea512d5-b1f6-4f87-8d42-92eba49fea60

The correct query string is constructed, but because of the POST not matching the switch, it gets discard.  I actually believe that the next line is _also_ wrong, as it only checks on "content" being null, which is _always_ the case.  I believe it should be checking for opt.content being null, but that's actually a different issue.

adelton, 2025-01-10:
Could you elaborate where POST with query parameters is exposed to users, or in what context you try to use that `Request.request`?

karlp, 2025-01-10:
This is the original problem, (at least in 19.07, might have been in other branches) obviously the link above to the "master" branch is worthless 5 years later.  https://github.com/openwrt/luci/blob/openwrt-19.07/modules/luci-base/htdocs/luci-static/resources/luci.js#L724-L737 

The same problematic code appears to have been preserved through 21.02, 22.03, 23.05, 24.10 and still into current master
*  https://github.com/openwrt/luci/blob/openwrt-21.02/modules/luci-base/htdocs/luci-static/resources/luci.js#L724-L736
* https://github.com/openwrt/luci/blob/openwrt-22.03/modules/luci-base/htdocs/luci-static/resources/luci.js#L724-L735
* https://github.com/openwrt/luci/blob/openwrt-23.05/modules/luci-base/htdocs/luci-static/resources/luci.js#L723-L736
* https://github.com/openwrt/luci/blob/openwrt-24.10/modules/luci-base/htdocs/luci-static/resources/luci.js#L680-L691
* https://github.com/openwrt/luci/blob/master/modules/luci-base/htdocs/luci-static/resources/luci.js#L680-L691

The problem was, and, presumably, from inspection, still is, that you can't use the L.Requests.request to make a POST with query parameters.... (as the title says)  Is that a good idea? no, but I don't make the rules.  The API document linked still exists, but it has the information on the format they expect with both query params _and_ POST on the second page: https://support.dexma.com/hc/en-gb/articles/360013772759-Using-the-insertion-API-to-introduce-data-in-a-gateway  Scroll down to "building the query" to see what they expected.

I worked around it at the time by using other parts of luci to request the "backend" to make the request via curl instead, because luci wouldn't allow it, hence me filing this ticket.

TL;DR: issue still certainly seems to be present.

adelton, 2025-01-10:
I was not asking where the `switch` command lives in master, I was able to determine that.

What I'm after is a step-by-step reproducer on the OpenWrt LuCI UI where the query parameters would be entered by the user (admin) and then not used.

karlp, 2025-01-10:
Obviously nothing would be in the UI, it doesn't work :)  I doubt there's anything in the existing Luci UI.  But it's an API provided to developers building applications, and it can't be used, hence the bug report.    For the API reference that was attempted to be used: See https://openwrt.github.io/luci/jsapi/LuCI.request.html  and in particular https://openwrt.github.io/luci/jsapi/LuCI.request.html#.RequestOptions where it mentions that the query can have strings appended, and the method can also be supplied, and it explicitly mentions POST.

If there's a separate repo for the APIs provided now and you want to move the bug there, cool, but at least when filed, bugs in this repo were for both the api and the "standard built in" applications, not just the standard built in applications only.

adelton, 2025-01-10:
I filed https://github.com/openwrt/luci/pull/7552 now, let's see what the reviewers think about that.

systemcrash, 2025-01-10:
I'll accept this since the semantics and usage of GET vs POST are no longer so black and white. It is assumed that developers know what they're doing, and that POST can invalidate a cached GET, but that's when we are mainly interacting with the luci front-end. 

As long as @karlp can verify that the proposed change does indeed do what he needs, then we can merge it.

karlp, 2025-01-10:
I no longer personally need this.  I had to work around it five years ago, so I filed the ticket to try and get the upstream APIs improved to be more useful.   I'm no longer employed on this either, so so I have very little personal motivation to try and setup a test for this.  If someone like @adelton wants to work on this, as they seem to, it should be pretty easy to demonstrate working non/working before/after though.