openwrt/luci #3628: Please extend the rollback time from 30 seconds to 5 minutes

mailinglists35, 2020-02-13:
I just wasted half an hour of my life with this stupid "feature": https://forum.openwrt.org/t/disabling-auto-rollback/16315

30 seconds is simply not enough when you need to change network configuration. We are humans, not speedy robots.

And the apply unckecked button never appeared, anyway!

castillofrancodamian, 2020-02-13:
That button is at 19.07 and master. If using an earlier version, change the `option rollback line '30'` to `option rollback '300'` in `/etc/config/luci`.

mailinglists35, 2020-02-13:
anyway the whole concept is totally wrong, until you extend by default the time to much more than 30 seconds, more new people will waste time because of this.

"apply unchecked" is also not easy to discover.

jow-, 2020-02-13:
Master and 19.07.1 offer an option to apply unchecked right away.

closed 2020-02-13 by jow-

castillofrancodamian, 2020-02-13:
Wouldn't it be better to implement in LuCI?