openwrt/luci #3643: Associated stations' host column shows only IPv6 addresses

bpados, 2020-02-18:
Dear LUCI team,

I'd like to report an issue when my 2nd router configured as a simple AP, then Associated stations' host column shows only IPv6 addresses instead of hostname (FQDN) and IPV4/IPv6 addresses in the bracket on this AP.
My primary router is configured as DHCP and DNS server.
Please see attached screenshot.
[image]
Best regards, bpados

jow-, 2020-02-18:
The information isn't available on the 2nd router.

closed 2020-02-18 by jow-

pgwipeout, 2025-09-29:
In case anyone comes across this, there is a solution. 

Allowing the ARP cache to become aware of the network devices is sufficient to make the data available to Luci. This can be done by installing arpscan, running it periodically and pinging all of the results.
Adding this to your cron jobs will run arpscan every fifteen minutes and update the arp cache. (Change br-lan to match your configuration)

` */15 * * * * arp-scan -qxlN -I br-lan | awk '{print $1}' | xargs fping -q -c1 `