openwrt/luci #3644: RFC: rework LuCI theme to Bootstrap with new design guidelines

aparcar, 2020-02-18:
Hi all, now that OpenWrt got a new [design guide](https://openwrt.org/docs/guide-graphic-designer/openwrt-logo) we should consider reworking the current theme to use adequate colors and maybe update the current bootstrap file ([1.4.0](https://github.com/openwrt/luci/blob/master/themes/luci-theme-bootstrap/htdocs/luci-static/bootstrap/cascade.css) vs [4.1.3](https://getbootstrap.com/docs/4.1/getting-started/introduction/)).

This also include renaming a *lot* of elements, however is surely worth the work! From my understanding it would mostly mean to rename classes created by the new JavaScript code.

Please share your opinion on this!

tmo26, 2020-02-20:
Good idea!

I just stumbled myself over a place where the new logo needs to be implemented: LuCI favicon.

- IS currently: [image]
- SHOULD BE in the future: [image]

Just as a reminder so it won't be forgotten.

ysc3839, 2020-02-23:
I found both bootstrap and material theme uses a highly modified css comparing to original bootstrap and mui. Is there any technological limit that we must modify css to make it works on luci? Can't we just use the original css? Because using original css makes authoring new theme easier.
@jow-

jow-, 2020-02-23:
> Can't we just use the original css?

The original CSS is too bloated. Unmodified Bootstrap 4.4.1 `bootstrap.min.css` is 159.5KB compared to 41.8 KB *unminified* in luci-theme-bootstrap.

ysc3839, 2020-02-23:
@jow- I'm not talking about size, those useless code can be deleted. The problem is, when I replaced `cascade.css` of bootstrap theme with the original 1.4.0 [bootstrap.min.css](https://getbootstrap.com/1.4.0/assets/css/bootstrap.min.css), the page become this:
[image]
So AFAIK, we have to modify css to make a new theme works on luci, which makes authoring new theme more difficult. I'm talking about can we use those css without making out own changes?

jow-, 2020-02-23:
Well you need to augment all LuCI code and replace CSS classes everywhere to match what bootstrap expects. The problem is that LuCI markup predates `luci-theme-bootstrap` quite some bit, so that bootstrap theme was adapted to LuCI, not the other way around.

ysc3839, 2020-02-23:
Well, I agree with you. Maybe we don't have a better solution. Implement the rendering code in every theme, that's a more complex work than modify css.

ysc3839, 2020-02-25:
I'm trying to do this by using Sass' `@extend`, for example:
```
.cbi-button {
  @extend .btn, .btn-sm;
}

.cbi-button-apply {
  @extend .btn-outline-primary;
}

.cbi-button-save {
  @extend .btn-outline-success;
}

.cbi-button-reset {
  @extend .btn-outline-danger;
}
```
But this will increase the size of final css. Is there any better way? How those themes are made?

Here's a screenshot of my work:
[image]

aparcar, 2020-02-25:
I'm not home in the CSS world but I remember a tool going trough HTML/JavaScript files and detect what CSS is actually used. Would that be a possible compression method for the "final" CSS?

aparcar, 2020-02-26:
@ysc3839 can you share your current work?

ysc3839, 2020-02-26:
@aparcar I'm considering creating a pull request. However If there's a lot of work to do, I may give up. I'm not familiar with Web front-end development. I'm just a bit interested.

jow-, 2020-03-23:
Please see #3780 for a custom theme implementation that does not require on bootstrap.

ysc3839, 2021-05-17:
@jow- @aparcar
I think the final solution for theming is to use Web Components.
Use `<cbi-button .../>` instead of `<button class="cbi-button" ...>`. And theme author can write it's own render function for the element.
https://developer.mozilla.org/en-US/docs/Web/Web_Components/Using_custom_elements

But this is not compatible with existing code, all code must changed to use the new `<cbi-button .../>` components.

jow-, 2021-05-18:
> But this is not compatible with existing code, all code must changed to use the new <cbi-button .../> components.

And that's why it is a non starter.