openwrt/luci #3648: LUCI Collect Error [v19.07 Snapshot]

solidus1983, 2020-02-18:
I have been trying to compile v19.07 and each time it has resulted in the same error

[code]
CCollected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci-ssl:
 * 	luci
 * opkg_install_cmd: Cannot install package luci-ssl.
[/code]

Luci is selected in the make menuconfig, I have complete wiped the Hyper-V and rebuilt it and still hit the same issues.

I normally don't have issues with luci during compile either normally.

[output.txt](https://github.com/openwrt/luci/files/4220779/output.txt)
[download.txt](https://github.com/openwrt/luci/files/4220781/download.txt)

jow-, 2020-02-18:
Cannot reproduce it here, nor can the buildbots. Maybe it is something specific to your system.

solidus1983, 2020-02-18:
Any ideas where to start looking? I have Adblock disabled when downloading from git and compiling.

I am using the .config from the last working compile too, could that be the issue?

jow-, 2020-02-18:
Did you run a `make defconfig` ?

solidus1983, 2020-02-18:
Yes I did. However will do a make clean, make defconfig and rerun the compile.

Thanks for your help.

closed 2020-02-19 by feckert