openwrt/luci #3651: Language and style error.

Holinc19, 2020-02-19:
When you install a theme, it would add an option in `Stlye`.But it does not remove the option while you uninstall the theme.
[image]
I have remove the theme of Material and OpenWrt, but it still can choose here. Is there something wrong?

feckert, 2020-02-20:
I think we have to update the theme Makefile in https://github.com/openwrt/luci/blob/master/themes/ with a postrm section to remove on remove the string from the config
```
config internal 'themes'
option Material
option OpenWrt
option Bootstrap
```

esershnr, 2024-11-12:
You can use ssh and type these;

```
uci -q delete luci.themes.Material
uci commit luci
```

adelton, 2025-01-11:
I believe this was implemented via https://github.com/openwrt/luci/pull/6191, so this issue can be closed as resolved.

closed 2025-01-12 by systemcrash