openwrt/luci #3652: LuCI: timezone information in uci is not updated/refreshed on sysupgrade

hmh, 2020-02-19:
The whole timezone configuration in OpenWRT is currently half-broken.

LuCI (mod-luci-admin-full) uses CBI to configure in UCI the timezone information:

 system.@system[0].zonename='Wolrd/timezone-name'
 system.@system[0].timezone='ATZ,-99'

Which is read in by an initscript of base-files, and the information from system.timezone is used to fill in /etc/TZ.

What is wrong:
* When timezone information (which is _not_ static for way too many places in this world) is updated in the relevant packages, it does not get automatically updated in UCI.  Since only the contents of UCI system.timezone matter (because it ends up in /etc/TZ), the effective timezone information will be left stale.

This results on systems entering/exiting DST incorrectly, even after the image (or packages) have been updated to the newest timezone database.

What should happen:
* on sysupgrade and relevant package update, the contents of system.zonename should be read, and if found, used to update the contents of system.timezone.

hnyman, 2020-02-20:
Uci itself has no timezone database. It only has that string in etc/config/system.

LuCI contains a small database of the timezone strings, and this list is shown to the user as the timezone selection drop-down. The list is up-to-date (IANA 2019c) in master, 19.07 and 18.06.

But there is no logic for automatically updating the timezone TZ string setting after database changes, as drafting a logic suitable for all possible changes and sysupgrades and restores etc. would not be achievable.

User can easily refresh the string by first switch timezone and then reselecting the correct one.

(And there aren't annually that many changes to timezones globally. Last year there was only three IANA database releases in the whole year)

closed 2020-02-20 by hnyman

hmh, 2020-02-21:
The LuCI module that sets it up should also take responsibility to update it through a script in uci-defaults or something like that.  The logic is simple:

"if the timezone *name* is set, and known to us, replace the timezone info to keep it current.  Otherwise, leave it alone."

It would likely have to be a lua module to keep it in sync with the tables that are actually used by LuCI, but uci-defaults takes lua scripts with the appropriate #! line just fine.

hmh, 2020-02-21:
On OpenWRT 18.06:

openwrt-luci/modules/luci-mod-admin-full/luasrc/model/cbi/admin_system/system.lua:
`o = s:taboption("general", ListValue, "zonename", translate("Timezone"))
o:value("UTC")

for i, zone in ipairs(zones.TZ) do
        o:value(zone[1])
end

function o.write(self, section, value)
        local function lookup_zone(title)
                for _, zone in ipairs(zones.TZ) do
                        if zone[1] == title then return zone[2] end
                end
        end

        AbstractValue.write(self, section, value)
        local timezone = lookup_zone(value) or "GMT0"
        self.map.uci:set("system", section, "timezone", timezone)
        fs.writefile("/etc/TZ", timezone .. "\n")
end
`

That's a (rather simple) timezone database no matter how much you'd like to call it otherwise, since it *is* setting the timezone information, not just its name :-(

On OpenWRT-19.07, things changed a great deal, but it *still* changes uci system.@system[0].timezone, so it *is* still a timezone database.

Only if LuCI ever touched system.@system[0].zonename, would it be possible to claim that it has no responsibility to keep UCI system.@system[0].zoneinfo up-to-date.

Please reopen the bug.  If nobody is willing to fix it right now, leave it open and maybe tag it "won't fix right now" or something.

jow-, 2020-02-21:
I don't want even more open bugs nobody is going to work on, so consider this one wontfix.

qcockroach, 2021-03-18:
Can uci change of specific file not only ones are under /etc/config ?