openwrt/luci #3653: Header topmenu shown logged out

johnnytolengo, 2020-02-19:
On commit r11655-a283b1788a the topmenu "<header>" is shown while no user logged in.

[image]

Any idea how to solve this issue?
Thank you.

jow-, 2020-02-20:
This is intentional.

closed 2020-02-20 by jow-

castillofrancodamian, 2020-02-26:
But it reveals device information. It should be removed or at least able to deactivate

feckert, 2020-02-26:
If I use the luci-theme-material i do not see the menu if i am not logged in!

jow-, 2020-02-26:
> But it reveals device information. It should be removed or at least able to deactivate

So does a `curl -v http://192.168.1.1/luci-static/resources/`. With the client side implementation it is not possible to hide what is available anymore, you can simply probe the URLs.

johnnytolengo, 2020-02-26:
I think it is not good to show that sensible information, the menu reveals what kind packages you have already installed and so on.

jow-, 2020-02-26:
Even if the menu is hidden you can still obtain that information easily, it provides zero security benefit.

johnnytolengo, 2020-02-26:
that is true, but I don't undestand what is the benefit/reason to show the menu while you are logged out.

jow-, 2020-02-26:
See #3563