openwrt/luci #3654: Sysupgrade via LuCI: "No space left on device"

iuncuim, 2020-02-20:
LuCI 19.07.1, device linkit7688 128MB RAM.
When I try download large (34,6 MB) sysupgrade file I get error:
```
28,No space left on device
Upload request failed: I/O failure while writing target file
```
Command _free_ output during update on 19.07:
```
root@OpenWrt:~# free
              total        used        free      shared  buff/cache   available
Mem:         125000       13332      101592         340       10076       82656
Mem:         125000       13836       99692        1736       11472       80756
Mem:         125000       13812       98152        3300       13036       79216
Mem:         125000       13788       96616        4860       14596       77680
Mem:         125000       13832       95072        6360       16096       76136
Mem:         125000       13792       93536        7936       17672       74600
Mem:         125000       13824       91936        9504       19240       73000
Mem:         125000       13816       90396       11052       20788       71460
Mem:         125000       13788       88888       12588       22324       69952
Mem:         125000       13828       87288       14148       23884       68352
Mem:         125000       13824       85692       15748       25484       66756
Mem:         125000       13796       84184       17284       27020       65252
Mem:         125000       13848       82584       18832       28568       63652
Mem:         125000       13804       81016       20444       30180       62084
Mem:         125000       13836       79420       22008       31744       60488
Mem:         125000       13820       77880       23564       33300       58952
Mem:         125000       13808       76344       25112       34848       57416
Mem:         125000       13844       74772       26648       36384       55844
Mem:         125000       13816       73232       28200       37952       54304
Mem:         125000       13788       71752       29724       39460       52824
Mem:         125000       13836       70152       31276       41012       51228
Mem:         125000       13840       68556       32868       42604       49632
Mem:         125000       13396       59780       42088       51824       40860
Mem:         125000       13304      101620         340       10076       82684
```

Command _free_ output during update on 18.06:
```
root@OpenWrt:~# free
             total       used       free     shared    buffers     cached
Mem:        125364      21876     103488        476       2152       6944
Mem:        125364      24656     100708       1268       2152       7736
Mem:        125364      25888      99476       2504       2152       8972
Mem:        125364      27148      98216       3756       2152      10224
Mem:        125364      28348      97016       4992       2152      11460
Mem:        125364      29640      95724       6216       2152      12684
Mem:        125364      30872      94492       7452       2152      13920
Mem:        125364      32104      93260       8688       2152      15156
Mem:        125364      33304      92060       9912       2152      16380
Mem:        125364      34536      90828      11152       2152      17620
Mem:        125364      35800      89564      12396       2152      18864
Mem:        125364      37060      88304      13632       2152      20100
Mem:        125364      38292      87072      14868       2152      21336
Mem:        125364      39524      85840      16096       2152      22564
Mem:        125364      40756      84608      17340       2152      23808
Mem:        125364      42016      83348      18580       2152      25048
Mem:        125364      43248      82116      19816       2152      26284
Mem:        125364      44480      80884      21060       2152      27528
Mem:        125364      45740      79624      22304       2152      28772
Mem:        125364      46912      78452      23524       2152      29992
Mem:        125364      48172      77192      24772       2152      31240
Mem:        125364      49404      75960      26004       2152      32472
Mem:        125364      50636      74728      27236       2152      33704
Mem:        125364      51900      73464      28484       2152      34952
Mem:        125364      53216      72148      29764       2152      36232
Mem:        125364      54504      70860      31040       2152      37508
Mem:        125364      55764      69600      32320       2152      38788
Mem:        125364      56996      68368      33560       2152      40028
Mem:        125364      57692      67672      34224       2152      40740
Mem:        125364      57692      67672      34224       2152      40740
Mem:        125364      57692      67672      34224       2152      40740
Mem:        125364      23568     101796        476       2152       6996

```

jow-, 2020-02-21:
This is related to how `cgi-io` handles file uploads. It first stages a temporary file in `/tmp`, then once completed, it copies that file to its final location. If the target file happens to be in `/tmp` too, like it is the case for firmware updates, it will occupy twice the space for a brief period of time, so 2 * 30MB which, together with opkg lists, likely overflow your 64MB tmpfs.

I'll see if I can implement an `O_TEMPFILE` based solution in cgi-io

jow-, 2020-02-22:
The linked PR has been merged, this should fix this issue.

closed 2020-02-22 by jow-