openwrt/luci #3660: network interface not deleted

mailyspamy, 2020-02-21:
If I delete wireless connection under "Network->Wireless" and want connect to a new one. Why I have to go to "Network->Interface" and delete the interface first. Than "save & apply". And after that I can connect to a new wireless network?

jow-, 2020-02-21:
Because logical interfaces and wireless networks are different entities in the OpenWrt configuration. Besides that, I don't know what you mean. I can simply delete and recreate wifi interfaces without issues here.

closed 2020-02-21 by jow-

mailyspamy, 2020-02-21:
I click on remove, than scan, i select new ssid and join network, but "Name of the new network" is "wwan2" and not "wwan". thats why i have to go to interfaces and delete the old wwan first.

jow-, 2020-02-21:
Ok, I understand. You can simply delete the `2` and reuse the empty `wwan` definition. Will change the logic to automatically reuse empty logical networks.

mailyspamy, 2020-02-21:
[image]
see picture. it is not possible

jow-, 2020-02-21:
Yep, noticed that too in my testing. After the linked commit, LuCI will automatically reuse leftover empty wwan interface. I decided against auto-deleting associated interfaces because it might be too dangerous with certain configurations as it would also delete related alias interfaces, routes etc.

commit 48e18fb8 (2020-02-21): luci-mod-network: wifi join: allow reclaiming empty wwan networks

Fixes: #3660
commit 62be6eff (2020-02-21): luci-mod-network: wifi join: allow reclaiming empty wwan networks

Fixes: #3660

(cherry picked from commit 48e18fb8acfdc661793582e4a907b10b9e8d0d43)