openwrt/luci #3661: BSSID missing

mailyspamy, 2020-02-21:
If I select a new network to connect, why is BSSID field empty and I have to manually write it?

jow-, 2020-02-21:
An option to use BSSID has been added. The rationale for not copying the BSSID is that many public networks are composed of multiple access points sharing the same SSID. Locking to a specific BSSID would break client roaming on such networks.

closed 2020-02-21 by jow-

mailyspamy, 2020-02-21:
i understand but in that case is possible to manualy delete the bssid, which is less work than search the bssid and write it manually. for example i use public networks in the oposite way. i want to connect to one bssid only. i hope that your option will be a sollution to my problem. thank you

commit 7ec007f9 (2020-02-21): luci-mod-network: wifi join: add option to lock bssid

Fixes: #3661
commit 4c8f0cdc (2020-02-21): luci-mod-network: wifi join: add option to lock bssid

Fixes: #3661

(cherry picked from commit 7ec007f926a9c966b3297085c4c46915372ba1dc)