openwrt/luci #3662: autorefresh

mailyspamy, 2020-02-21:
If I scan and search new network, why is it repeatedly refreshing? Because of issue#3661 I have to manually select the BSSID and copy it. But it is almost impossible because of the refresh rate. Can the refresh be disabled?

jow-, 2020-02-21:
The refresh is required because some radios, especially 5 GHz ones require many scan cycles to discover all neighboring networks.

closed 2020-02-21 by jow-

mailyspamy, 2020-02-21:
and is it possible to let me decide when i want to refresh the site?

jow-, 2020-02-21:
Since the linked commit yes.

mailyspamy, 2020-02-21:
great thank you. i did not see before posting.

mailyspamy, 2020-03-06:
after pressing "stop refresh" it refreshes once more time and than stops. its like the button stops scheduling refresh, but the last scheduled refresh happens.

commit a2fe3d9f (2020-02-21): luci-mod-network: wifi scan: add button to start/stop refreshes

Fixes: #3662
commit 29c29a6b (2020-02-21): luci-mod-network: wifi scan: add button to start/stop refreshes

Fixes: #3662

(cherry picked from commit a2fe3d9fc4f967b9d79346f358aaaf12eeb1e040)