openwrt/luci #3670: luci-mod-network: Does not display acurate wireless status

aaegitt, 2020-02-23:
When multiple SSIDs are configured on a radio, only the first enabled SSID has the correct information displayed in the Wireless Overview and Associated Station sections.  All other SSIDs show "Wireless is not associated" regardless if any stations are connected.  Using iwinfo  assoclist will show connected stations, but these do not appear in the Associated Station list.

I found others have mentioned this problem in the Openwrt forum:
 https://forum.openwrt.org/t/davidc502-wrt1200ac-wrt1900acx-wrt3200acm-wrt32x-builds/15839

This behavior was not seen in my previous build from 4 months ago.  I am now running:
LuCI Master (git-20.052.76652-4322cae) / OpenWrt SNAPSHOT r12261-d1130ad265

jow-, 2020-02-23:
Please first test whether the problem persists after

```
make package/rpcd-mod-iwinfo/clean
make package/rpcd-mod-luci/clean
make package/iwinfo/clean
```

aaegitt, 2020-02-23:
I executed the above as requested with the following modification:

make package/rpcd-mod-iwinfo/clean  - resulted in Error 2
changed to:
make package/rpcd/clean - completed without error

The new build has the same problem with inaccurate info.  I also tried a make dirclean with the same result.  Let me know if you have anymore suggestions.

jow-, 2020-02-24:
Unfortunately I cannot reproduce the issue locally. Please provide the complete output of `ubus call luci-rpc getWirelessDevices`

aaegitt, 2020-02-24:
As requested.

[ubus output.txt](https://github.com/openwrt/luci/files/4246352/ubus.output.txt)

jow-, 2020-02-26:
Please provide the output of `ubus call network.wireless status` as well

aaegitt, 2020-02-26:
As requested.

[ubus wireless output.txt](https://github.com/openwrt/luci/files/4255874/ubus.wireless.output.txt)

jow-, 2020-02-26:
The problem seems related to `netifd`. It does not report the ifnames for the second and third VAP, that's why LuCI cannot correlate the iwinfo runtime data to it.

kofec, 2020-02-26:
I also see this issue XIAOMI MINI hardware for 5G interface(MT76) 
for 2,4GHz it is OK(rt2x00)

jow-, 2020-02-26:
Seems like a regression in OpenWrt master.

aaegitt, 2020-03-02:
Problem was resolved by resetting config back to default then manually re-entering information.

closed 2020-03-02 by aaegitt