openwrt/luci #3671: 19.07.0 and 19.07.1 alaway show"Loading package information…",can not show package list properly

be-engineer, 2020-02-23:
My device is tp-link wdr7500. Each time I open the software list does not display properly,click on the Update lists situations remain.
[image]

be-engineer, 2020-02-23:
it's my config fault.closed!

closed 2020-02-23 by be-engineer

bordeux, 2021-03-20:
@be-engineer what was the problem exactly? Also im affected  with this problem in fresh installation

leleb, 2021-03-20:
[image]

me too, fresh install in a esxi x86 vm. After "update lists" it is stuck with "loading package information"

bordeux, 2021-03-20:
Right now i know, it is not backend issue, but Frontend issue :)

[image]

leleb, 2021-03-20:
updated to 19.07.7 now working