openwrt/luci #3673: Not showing IPv4 upstream in Overview

brianjmurrell, 2020-02-23:
On my router I have:

```
Powered by LuCI openwrt-19.07 branch (git-20.053.76840-087e6ac) / OpenWrt 19.07.1 r10911-c155900f66
```

I have two upstream connections (IOW, 2 `wan` interfaces) however in the main Overview page, it's not showing one of my IPv4 upstreams.  It's showing the IPv6 upstream for that interface, just not the IPv4 upstream:

[image]

That pppoe-wan1 interface's addresses:

```
13: pppoe-wan1: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP> mtu 1492 qdisc fq_codel state UNKNOWN group default qlen 3
    link/ppp 
    inet [redacted valid IPv4 address] peer [redacted valid IPv4 address]/32 scope global pppoe-wan1
       valid_lft forever preferred_lft forever
    inet6 [redacted valid IPv6 address]/64 scope global dynamic noprefixroute 
       valid_lft 2591736sec preferred_lft 604536sec
    inet6 fe80::c7e:570c:b753:9c74/10 scope link 
       valid_lft forever preferred_lft forever
```

The IPv4 address of the above interface is missing from the upstream list in the Overview, as you can see.

jow-, 2020-02-23:
Please report the output of `ubus call network.interface dump`

brianjmurrell, 2020-02-23:
```
{
	"interface": [
		{
			"interface": "guest",
			"up": true,
			"pending": false,
			"available": true,
			"autostart": true,
			"dynamic": false,
			"uptime": 17728,
			"l3_device": "br-guest",
			"proto": "static",
			"device": "br-guest",
			"updated": [
				"addresses"
			],
			"metric": 0,
			"dns_metric": 0,
			"delegation": true,
			"ipv4-address": [
				{
					"address": "192.168.101.253",
					"mask": 24
				},
				{
					"address": "192.168.101.254",
					"mask": 24
				}
			],
			"ipv6-address": [
				
			],
			"ipv6-prefix": [
				
			],
			"ipv6-prefix-assignment": [
				
			],
			"route": [
				
			],
			"dns-server": [
				
			],
			"dns-search": [
				
			],
			"neighbors": [
				
			],
			"inactive": {
				"ipv4-address": [
					
				],
				"ipv6-address": [
					
				],
				"route": [
					
				],
				"dns-server": [
					
				],
				"dns-search": [
					
				],
				"neighbors": [
					
				]
[... 703 lines cut ...]
				],
				"neighbors": [
					
				]
			},
			"data": {
				"passthru": "[redacted]"
			}
		},
		{
			"interface": "wan6",
			"up": false,
			"pending": true,
			"available": true,
			"autostart": true,
			"dynamic": false,
			"proto": "dhcpv6",
			"device": "eth0.2",
			"data": {
				
			}
		}
	]
}
```

jow-, 2020-02-23:
LuCI only displays interfaces with active default routes as upstream, so it works as designed.

closed 2020-02-23 by jow-

brianjmurrell, 2020-02-24:
How is it determining an active default route?

Surely you can understand how this is flawed in a case where you have multiple upstreams in an active/backup scenario.  Don't you think in that case, both upstreams should be displayed?

jow-, 2020-02-24:
> How is it determining an active default route?

By scanning the ubus interface status for interfaces with default routes. In your case, only one IPv4 interface has an active default route entry so only one is displayed.

brianjmurrell, 2020-02-24:
But surely you can understand how this is flawed in a case where you have multiple upstreams in an active/backup scenario. Don't you think in that case, both upstreams should be displayed?

jow-, 2020-02-24:
No I can't understand it. The purpose of these boxes has always been to displays the interface currently being the gateway to the internet.

brianjmurrell, 2020-02-24:
FWIW, they both do have a default route plumbed:

```bash
# ip route ls | grep default
# ip rule ls
0:	from all lookup local 
999:	from all lookup main 
...
32765:	from all lookup balance 
32767:	from all lookup default 
# ip route ls table balance
default via 7.3.7.254 dev eth0.2 
# ip route ls table default
default dev pppoe-wan1 scope link metric 2 
```

But of course routing table `balance` is used before routing table `default`, and only when routing table `balance` is removed from the routing rules (because it's upstream has gone down) does routing table `default` get utilized.

> The purpose of these boxes has always been to display the interface currently being the gateway to the internet.

Which makes sense in a single-wan situation, I agree.  But in a multi-wan configuration do you not think it would be more useful to show all functioning upstream interfaces, with perhaps an additional indication as to which are active and which are not?  Maybe the colour of the box border and/or an Active: Yes/No indicator, etc.

After-all, in a multi-wan situation, even when I have configured one of the connections to be purely backup, I would still want to know that it's up and ready to be the active connection in the case of the currently active connection going down, wouldn't I?

To be clear, I am talking about an RFE here, not a bug in the way it's currently working.

jow-, 2020-02-24:
> But in a multi-wan configuration do you not think it would be more useful to show all functioning upstream interfaces, with perhaps an additional indication as to which are active and which are not? Maybe the colour of the box border and/or an Active: Yes/No indicator, etc.

This is exactly what luci-app-mwan3 implements.

> To be clear, I am talking about an RFE here, not a bug in the way it's currently working.

I have no intentions to implement this in base LuCI.

brianjmurrell, 2020-02-24:
> This is exactly what luci-app-mwan3 implements.

Which only works if one uses `[luci-app-]mwan3` for one's multi-wan implementation.  :frowning:

Does it work by altering the Overview as discussed here or does it leave that alone and add a new view somewhere else of available and active interfaces?