openwrt/luci #3674: Don't show "non-information" for uninstalled components

brianjmurrell, 2020-02-23:
When a component is not installed, such as dnsmasq, it's wasteful and distracting to show "non-information" about it.  I.e.:

[image]

There are no leases because this router doesn't run dnsmasq nor does it manage DHCP with `odhcpd`.

Showing that there are no leases is not useful in such a scenario.

I am sure there are other assumptions about installed components in various Luci pages also.  I hope this issue can encompass all of such, but if not, at least the issue about leases in the Overview page could be covered.

```
Powered by LuCI openwrt-19.07 branch (git-20.053.76840-087e6ac) / OpenWrt 19.07.1 r10911-c155900f66
```

closed 2020-02-23 by jow-

brianjmurrell, 2020-02-24:
Cool.  Can't wait to see this in action.  I have noticed that luci gets updated a lot in the lifecycle of point release.  Is this the sort of change that we are likely to see in 19.07.x or will this wait for some future major release update?

jow-, 2020-02-24:
Undecided.

brianjmurrell, 2020-03-11:
So I see that this [landed in 19.07](https://github.com/openwrt/luci/commit/02c68f2838197e4eff63c61a89e60164ba3faa1f) and I confirmed that my `/www/luci-static/resources/view/status/include/40_dhcp.js` has the changes in it:

```javascript
'use strict';'require rpc';'require network';var callLuciDHCPLeases=rpc.declare({object:'luci-rpc',method:'getDHCPLeases',expect:{'':{}}});return L.Class.extend({title:'',load:function(){return Promise.all([L.resolveDefault(callLuciDHCPLeases(),{}),network.getHostHints()]);},renderLeases:function(data){var leases=Array.isArray(data[0].dhcp_leases)?data[0].dhcp_leases:[],leases6=Array.isArray(data[0].dhcp6_leases)?data[0].dhcp6_leases:[],machints=data[1].getMACHints(false);var table=E('div',{'class':'table'},[E('div',{'class':'tr table-titles'},[E('div',{'class':'th'},_('Hostname')),E('div',{'class':'th'},_('IPv4-Address')),E('div',{'class':'th'},_('MAC-Address')),E('div',{'class':'th'},_('Leasetime remaining'))])]);cbi_update_table(table,leases.map(function(lease){var exp;if(lease.expires===false)
exp=E('em',_('unlimited'));else if(lease.expires<=0)
exp=E('em',_('expired'));else
exp='%t'.format(lease.expires);return[lease.hostname||'-',lease.ipaddr,lease.macaddr,exp];}),E('em',_('There are no active leases')));var table6=E('div',{'class':'table'},[E('div',{'class':'tr table-titles'},[E('div',{'class':'th'},_('Host')),E('div',{'class':'th'},_('IPv6-Address')),E('div',{'class':'th'},_('DUID')),E('div',{'class':'th'},_('Leasetime remaining'))])]);cbi_update_table(table6,leases6.map(function(lease){var exp;if(lease.expires===false)
exp=E('em',_('unlimited'));else if(lease.expires<=0)
exp=E('em',_('expired'));else
exp='%t'.format(lease.expires);var hint=lease.macaddr?machints.filter(function(h){return h[0]==lease.macaddr})[0]:null,host=null;if(hint&&lease.hostname&&lease.hostname!=hint[1]&&lease.ip6addr!=hint[1])
host='%s (%s)'.format(lease.hostname,hint[1]);else if(lease.hostname)
host=lease.hostname;else if(hint)
host=hint[1];return[host||'-',lease.ip6addrs?lease.ip6addrs.join(' '):lease.ip6addr,lease.duid,exp];}),E('em',_('There are no active leases')));return E([E('h3',_('Active DHCP Leases')),table,E('h3',_('Active DHCPv6 Leases')),table6]);},render:function(data){if(L.hasSystemFeature('dnsmasq')||L.hasSystemFeature('odhcpd'))
return this.renderLeases(data);return E([]);}});
```
 but I am still seeing the empty blocks:

[image]

and I don't have either `dnsmasq` nor `odhcpd` installed:

```
# opkg list-installed | grep -e odhcpd -e dnsmasq
odhcpd-ipv6only - 2019-12-16-e53fec89-3
```

jow-, 2020-03-11:
odhcpd-ipv6only triggers it.

brianjmurrell, 2020-03-11:
Pity.  I guess this is a pretty coarse-grained guess at whether the node is actually leasing addresses and not really considering that the functionality might be present (in the case of `odhcpd` because you want other functionality from the package) but disabled.

commit 4efc7cbb (2020-02-23): luci-mod-status: hide DHCP status if dnsmasq/odhcpd aren't present

Fixes: #3674
commit 02c68f28 (2020-02-23): luci-mod-status: hide DHCP status if dnsmasq/odhcpd aren't present

Fixes: #3674

(cherry picked from commit 4efc7cbbda86280343683a7d1c2563233c499550)