openwrt/luci #3682: Status Overview free memory

bjlockie, 2020-02-26:
The used memory is the white area?

[image]

I think it is backwards, the blue should be how much memory is in use.
"Free" should say "Free/In Use" and then have the blue bar be a gauge. :slight_smile:
I don't see a need for "Total Available", one bar can do the job of both.

feckert, 2020-02-26:
Feel free to open a pullrequest for this change

bjlockie, 2020-02-26:
Does a pull request mean I change it and then submit it?
How do I test the changes on anything other than my router?

feckert, 2020-02-26:
> Does a pull request mean I change it and then submit it?

Yes

> Does a pull request mean I change it and then submit it? How do I test the changes on anything other than my router?

Thats the way to go or you could also build a vmdk for virtualbox

PR #3694 (2020-03-01): luci-base: Show used memory instead of Free
This makes sure solid/blue bar shows the used memory. Fix #3682
closed 2020-03-01 by jow-