openwrt/luci #3689: Can't go into the luci web manage page when device runs a long time , For me the time is 42days

truelv, 2020-02-27:
**Below shows the error message.**  
/usr/lib/lua/luci/dispatcher.lua:562: Failed to execute function dispatcher target for entry '/'.
The called action terminated with an exception:
/usr/lib/lua/luci/dispatcher.lua:562: Failed to execute firstchild dispatcher target for entry '/admin'.
The called action terminated with an exception:
/usr/lib/lua/luci/dispatcher.lua:562: Failed to execute function dispatcher target for entry '/admin/status'.
The called action terminated with an exception:
/usr/lib/lua/luci/dispatcher.lua:562: Failed to execute template dispatcher target for entry '/admin/status/overview'.
The called action terminated with an exception:
/usr/lib/lua/luci/template.lua:97: Failed to load template 'admin_status/index'.
Error while parsing template '/usr/lib/lua/luci/view/admin_status/index.htm':
No such file or directory
stack traceback:
	[C]: in function 'assert'
	/usr/lib/lua/luci/dispatcher.lua:562: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:306: in function </usr/lib/lua/luci/dispatcher.lua:305>

**I use Telnet to go into the file system** and see index.htm is only in /overlay/usr/lib/lua/luci/view/admin_status/index.htm 
/rom/usr/lib/lua/luci/view/admin_status/index.htm
and they are not abnormal 
treebear@Witown:/overlay/usr/lib/lua/luci/view/admin_status# ls -l
lrwxrwxrwx    1 treebear root            18 Feb 25 10:33 index.htm -> (overlay-whiteout)
treebear@Witown:/overlay/usr/lib/lua/luci/view/admin_status# ls -l
lrwxrwxrwx    1 treebear root            18 Feb 25 10:33 index.htm -> (overlay-whiteout)

Then **I look at Top information**, it goes well because cpu and memory has lots of idle resource:
Mem: 134908K used, 121092K free, 0K shrd, 6116K buff, 45960K cached
CPU:   0% usr   2% sys   0% nic  97% idle   0% io   0% irq   0% sirq
Load average: 0.04 0.12 0.13 2/80 17638
  PID  PPID USER     STAT   VSZ %VSZ %CPU COMMAND
17638 14983 treebear R     1512   1%   2% top
24433  1882 treebear S    57108  22%   0% /usr/bin/ppcloud
 1465     1 treebear S    10060   4%   0% /usr/sbin/dnsmasq -C /var/etc/dnsmasq
 1827     1 treebear S     4224   2%   0% /usr/sbin/lighttpd -f /etc/lighttpd/l
 1201     1 treebear S     1544   1%   0% /sbin/netifd
 1924     1 treebear S     1528   1%   0% /sbin/syslogd -l 7 -O /etc/log/system
 3636  1831 treebear S     1520   1%   0% {update_cur_stat} /bin/sh /usr/sbin/u
14983  1759 treebear S     1520   1%   0% -ash
 1759     1 treebear S     1516   1%   0% /usr/sbin/telnetd
 1804     1 treebear S     1512   1%   0% /usr/sbin/ntpd -n -u 86400 -r 600 -p
  471     1 treebear S     1512   1%   0% /bin/ash --login
 1882     1 treebear S     1512   1%   0% {witownapp.sh} /bin/sh /usr/sbin/wito
 1927     1 treebear S     1508   1%   0% /usr/sbin/crond start
 1348  1201 treebear S     1508   1%   0% udhcpc -R -p /var/run/udhcpc-eth0.5.p
 1925     1 treebear S     1504   1%   0% /sbin/klogd
 1831     1 treebear S     1504   1%   0% {runing_status_m} /bin/sh /usr/sbin/r
17637  3636 treebear S     1500   1%   0% sleep 10
    1     0 treebear S     1360   1%   0% /sbin/procd
 1782     1 treebear S     1084   0%   0% /sbin/mountd -f
 923     1 treebear S      980   0%   0% arp_broadcast lan

**I try to use rm -rf /tmp/luci-* but it seems useless.** 

Does this issue comes from openwrt filesystem or it is luci's bug ?

System info:
Linux  3.10.34  mips
lua-5.1.5
openwrt CC

ysc3839, 2020-02-27:
Maybe it is a bug, but this version of OpenWrt is 5 years old and not supported. You should try the latest version and see if this bug still exist.

truelv, 2020-02-27:
But our customized system is based on openwrt CC, this will cost too much to change to another version. @ysc3839 . Thx for your reply.

feckert, 2020-02-27:
I would also suggest going to a new stable openwrt stable.
I remember having the same problem with 18.06.
a `/etc/init.d/uhttpd restart` fixed the problem on my system.

truelv, 2020-02-27:
You said you have the same problem with 18.06. So it means that the newest openwrt version can't work well then probably no help to me if I go to a new stable openwrt.

I tried /etc/init.d/lighttpd restart and failed.   Thanks for your reply ! @feckert

jow-, 2020-02-27:
Not a LuCI issue. Something deleted essential files on your system.

closed 2020-02-27 by jow-