openwrt/luci #3709: luci-mod-network: Interface wrongly re-assigned when port VLAN membership changed

Valentin-N, 2020-03-01:
## Summary
Adding an untagged external port to a VLAN results in the interface for that VLAN getting re-assigned to `eth0`, breaking connectivity for that interface.

## Scenario
VLAN100 is tagged on the CPU port and one or more tagged/untagged external ports are already assigned to VLAN100 and things work as expected. Adding one more untagged external port to VLAN100 and applying the changes results in the below message popping under the "waiting for config to be applied" dialogue:
```
Interface "lan" device auto-migrated from "eth0.100" to "eth0"
```
The result is that the `lan` interface stops working due to the incorrect re-assignment.

## Details
I was able to trace this feature back to https://github.com/openwrt/luci/commit/aa2b1cdc2cc56d015851af4a62169bfb85739890, where the author comments this functionality is only supposed to be triggered when assigning tagged/untagged CPU ports to prevent loss of management access. With https://github.com/openwrt/luci/commit/82743b3bd4b4603f2c6b85566d07182cb0f1ce52 it seems that the new implementation gets triggered for external ports as well, resulting in the incorrect behavior detailed above.

I can replicate this on:
```
LuCI openwrt-19.07 branch (git-20.029.45734-adbbd5c) / OpenWrt 19.07.1 r10911-c155900f66
```

I can NOT replicate this on:
```
LuCI openwrt-18.06 branch (git-20.029.49294-41e2258) / OpenWrt 18.06.7 r7976-ca47026b7d
```

As a side note, OpenWRT already automatically rolls back to a known working config when management access is lost. Given this, I don't see what is gained by having something alter the interface config with no user confirmation beforehand.

jow-, 2020-03-02:
Please provide your `/etc/board.json` and `/etc/config/network` contents before the error occurs. I am unable to reproduce this locally.

Valentin-N, 2020-03-02:
Attached (just remove `.log` from file name):
* [board.json.log](https://github.com/openwrt/luci/files/4273964/board.json.log)
* [network.log](https://github.com/openwrt/luci/files/4273965/network.log)

jow-, 2020-03-02:
Thank you, the issue was a reversed internal port ordering.

closed 2020-03-02 by jow-

farfade, 2021-04-20:
I apologize if my issue is different (maybe the issue is myself), but can someone tell me why the openwrt UI says it auto-reassign when I change LAN 4 on VLAN 3 from tagged to untagged ?

[image]

Cheers,

farfade

commit 1c547d36 (2020-03-02): luci-mod-network: switch.js: fix internal port ordering

Fixes: #3709
commit 70b5982b (2020-03-02): luci-mod-network: switch.js: fix internal port ordering

Fixes: #3709

(cherry picked from commit 1c547d36b545e79838acca021767071ae95ca5c9)