openwrt/luci #3734: Enhancement:  status/overview/wireless associated stations

cavanaug, 2020-03-09:
It would be really nice if the associated stations list would show more than just the IP address for the host.   With ipv6 addresses its sort of hopeless to look at it and ascertain what host it is.

Would it be possible to perform a reverse dns lookup on the ip and include the hostname as well in the status page?   It would help readability significantly...

howard0su, 2020-03-10:
It is there (19.07). It shows the hostname name with ipv4 and ipv6 address.

jow-, 2020-03-11:
Same for me.

closed 2020-03-11 by jow-

cavanaug, 2020-03-11:
I believe the defect may only exhibit itself when using openwrt in AP mode.   I use OpenWrt in AP mode for 3 APs and it never shows the hostname.   It will occasionally show the IP address.   See below

[image]

jow-, 2020-03-11:
That information simply isn't available in pure AP mode. Since traffic is bridged, no ARP / IPv6 neighbor discovery ever happens.

cavanaug, 2020-03-11:
In cases where the IP address is available how is the hostname determined?
  I can do a reverse dns lookup and it gives the name.