openwrt/luci #3736: Unable to open Diagnostics page

kudlav, 2020-03-09:
> Failed to execute template dispatcher target for entry '/admin/network/diagnostics'. The called action terminated with an exception: /usr/lib/lua/luci/template.lua:74: Failed to load template 'admin_network/diagnostics'. Error while parsing template '/usr/lib/lua/luci/view/admin_network/diagnostics.htm': No such file or directory stack traceback: 	[C]: in function 'error' 	/usr/lib/lua/luci/template.lua:74: in function '__init__' 	/usr/lib/lua/luci/util.lua:65: in function 'Template' 	/usr/lib/lua/luci/template.lua:27: in function 'render' 	/usr/lib/lua/luci/dispatcher.lua:920: in function </usr/lib/lua/luci/dispatcher.lua:919>

Model: TP-Link TL-WR1043N/ND v1
Architecture: Atheros AR9132 rev 2
Firmware Version: OpenWrt 19.07.2 r10947-65030d81f3 / LuCI openwrt-19.07 branch git-20.067.35699-cf18bdd
Kernel Version: 4.14.171

jow-, 2020-03-14:
`rm -f /tmp/luci-indexcache` or reboot.

closed 2020-03-14 by jow-

db260179, 2020-04-13:
And rm -rf /tmp/luci-modulecache
is required

kudlav, 2020-04-13:
Reboot solved the problem for me, but it's inconvenience to see the failure.