openwrt/luci #3740: Error canceling load quickly

castillofrancodamian, 2020-03-10:
It appears throughout the LuCI interface when I do I do that action. I have snapshots.

[image]

jow-, 2020-03-11:
Yes, don't do that then :) If you cancel page loading midway, related XHR requests will be aborted as well, no way to change that.

closed 2020-03-11 by jow-

dibdot, 2020-03-11:
@jow- May I ask why do we report this "error" to the user at all? The popup comes up for a few seconds and then automatically disappears, cause the user already clicked another top level menu option. Maybe there are other use cases where this error popup makes more sense, I don't know.

Thanks!

jow-, 2020-03-11:
Good question, I guess we can simply hide this particular case.

castillofrancodamian, 2020-03-11:
That's the problem.  That appears all in time.  I forgot to clarify.