openwrt/luci #3743: luci-app-statistics: no graphs displayed w/ collectd-mod-apcups

aaegitt, 2020-03-11:
When APC UPS tab is selected, nothing is displayed except "Loading data..."  RRD files are being updated every minute.

OpenWrt SNAPSHOT r12458-6e80df5e33 / LuCI Master git-20.064.70038-6c167ea

jow-, 2020-03-14:
Can you provide me with an XML export of the apcups RRD files?

aaegitt, 2020-03-15:
As requested.
[charge.xml.txt](https://github.com/openwrt/luci/files/4333665/charge.xml.txt)
[percent-load.xml.txt](https://github.com/openwrt/luci/files/4333666/percent-load.xml.txt)
[timeleft.xml.txt](https://github.com/openwrt/luci/files/4333667/timeleft.xml.txt)
[voltage-battery.xml.txt](https://github.com/openwrt/luci/files/4333668/voltage-battery.xml.txt)
[voltage-input.xml.txt](https://github.com/openwrt/luci/files/4333669/voltage-input.xml.txt)

commit 12e1d941 (2020-03-16): luci-app-statistics: fix APC UPS graph rendering

Fixes: #3743
closed 2020-03-16 by jow-