openwrt/luci #3744: luci is blocked when invoking shell commands

truelv, 2020-03-11:
**platform**：
openwrt CC / mips ( 4 cpu core)

**Repetition frequency**：
sometimes happen 

**wrong lines**：
luci/modules/admin-full/luasrc/view/admin_status/index.html  line97   
mac = luci.sys.exec("ifconfig eth0 | awk '/HWaddr/{print $5}'")

**PS info：**
  15 treebear     0 SW   [ksoftirqd/2]
   16 treebear     0 SW   [kworker/2:0]
   17 treebear     0 SW<  [kworker/2:0H]
   18 treebear     0 SW   [migration/3]
   19 treebear     0 SW   [ksoftirqd/3]
   20 treebear     0 SW   [kworker/3:0]
   21 treebear     0 SW<  [kworker/3:0H]
   22 treebear     0 SW<  [khelper]
  100 treebear     0 SW<  [writeback]
  102 treebear     0 SW<  [bioset]
  104 treebear     0 SW<  [kblockd]
  117 treebear     0 SW   [khubd]
  144 treebear     0 RW   [kworker/0:1]
  150 treebear     0 SW   [kswapd0]
  194 treebear     0 SW   [fsnotify_mark]
  266 treebear     0 SW   [kworker/3:1]
  267 treebear     0 SW   [kworker/2:1]
  268 treebear     0 SW   [kworker/1:1]
  280 treebear     0 SW<  [deferwq]
  382 treebear     0 SWN  [jffs2_gcd_mtd6]
  461 treebear   888 S    /sbin/ubusd
  470 treebear   768 S    /sbin/askfirst ttyS0 /bin/ash --login
  912 treebear     0 SW<  [rpciod]
  933 treebear     0 SW<  [nfsiod]
 1164 treebear  1504 S    /sbin/klogd
 1165 treebear  1528 S    /sbin/syslogd -l 7 -O /etc/log/system.log -s 1000 -b 10 -U /tmp/user.log
 1166 treebear     0 SW   [kworker/0:2]
 1202 treebear  1544 S    /sbin/netifd
 1353 treebear  1508 S    udhcpc -R -p /var/run/udhcpc-eth0.5.pid -s /lib/netifd/dhcp.script -f -t 0 -i eth0.5 -H Witown -V #AP:Tre
 1466 treebear  1208 S    /usr/sbin/dnsmasq -C /var/etc/dnsmasq.conf -k -u treebear
 1468 treebear     0 SW<  [kworker/1:1H]
 1486 treebear     0 SW   [RtmpCmdQTask]
 1487 treebear     0 SW   [RtmpWscTask]
 1488 treebear     0 SW   [RtmpMlmeTask]
 1650 treebear     0 SW   [RtmpCmdQTask]
 1651 treebear     0 SW   [RtmpWscTask]
 1652 treebear     0 SW   [RtmpMlmeTask]
 1745 treebear  1508 S    /usr/sbin/telnetd
 1769 treebear  1084 S    /sbin/mountd -f
 1791 treebear  1508 S    /usr/sbin/ntpd -n -u 86400 -r 600 -p 0.cn.pool.ntp.org -p 1.cn.pool.ntp.org -p 2.cn.pool.ntp.org -p 3.cn.
 1815 treebear  4032 S    /usr/sbin/lighttpd -f /etc/lighttpd/lighttpd.conf
 1819 treebear  1504 S    {runing_status_m} /bin/sh /usr/sbin/runing_status_monitor
 1833 treebear  1520 S    {update_cur_stat} /bin/sh /usr/sbin/update_cur_status
 1870 treebear  1508 S    {witownapp.sh} /bin/sh /usr/sbin/witownapp.sh
 1905 treebear   980 S    arp_broadcast lan
 1911 treebear  1508 S    /usr/sbin/crond start
 1930 treebear 37408 S    /usr/bin/ppcloud
 2005 treebear     0 SW   [kworker/u8:2]
 6786 treebear  1520 S    -ash
15051 treebear  3320 S    {luci} /usr/bin/lua /www/cgi-bin/luci    **//these three lines below are bocked one bye one** 
15120 treebear  1512 S    sh -c ifconfig eth0 | awk '/HWaddr/{print $5}'
15122 treebear  1512 R    sh -c ifconfig eth0 | awk '/HWaddr/{print $5}'
22209 treebear  1500 S    sleep 10
22214 treebear  1508 R    ps w

**Top info：**
Mem: 83636K used, 172364K free, 0K shrd, 5748K buff, 22196K cached
CPU:   4% usr  29% sys   0% nic  50% idle   0% io   0% irq  15% sirq
Load average: 1.23 1.33 1.45 2/77 24291
  PID  PPID USER     STAT   VSZ %VSZ %CPU COMMAND
15122 15120 treebear R     1512   1%  25% sh -c ifconfig eth0 | awk '/HWaddr/{p    **//wrong line 1**
    3     2 treebear SW       0   0%   7% [ksoftirqd/0]
 1930  1870 treebear S    37412  15%   2% /usr/bin/ppcloud
24289  6786 treebear R     1512   1%   2% top
 1815     1 treebear S     4032   2%   0% /usr/sbin/lighttpd -f /etc/lighttpd/l
15051  1815 treebear S     3320   1%   0% {luci} /usr/bin/lua /www/cgi-bin/luci  **//wrong line 3**
 1202     1 treebear S     1544   1%   0% /sbin/netifd
 1165     1 treebear S     1528   1%   0% /sbin/syslogd -l 7 -O /etc/log/system
 1833  1819 treebear S     1520   1%   0% {update_cur_stat} /bin/sh /usr/sbin/u
 6786  1745 treebear S     1520   1%   0% -ash
15120 15051 treebear S     1512   1%   0% sh -c ifconfig eth0 | awk '/HWaddr/{p  **//wrong line 2**
 1745     1 treebear S     1508   1%   0% /usr/sbin/telnetd
 1911     1 treebear S     1508   1%   0% /usr/sbin/crond start
 1353  1202 treebear S     1508   1%   0% udhcpc -R -p /var/run/udhcpc-eth0.5.p
 1791     1 treebear S     1508   1%   0% /usr/sbin/ntpd -n -u 86400 -r 600 -p
 1870     1 treebear S     1508   1%   0% {witownapp.sh} /bin/sh /usr/sbin/wito
 1164     1 treebear S     1504   1%   0% /sbin/klogd
 1819     1 treebear S     1504   1%   0% {runing_status_m} /bin/sh /usr/sbin/r
24282  1833 treebear S     1500   1%   0% sleep 10
  1        0 treebear S     1360   1%   0% /sbin/procd
  
wrong line 3 is blocked by wrong line 2 and wrong line 2 is blocked by wrong line 1.  
 
Hopes for your reply soon.
Thanks a lot.

truelv, 2020-03-11:
[image]
[image]

hnyman, 2020-03-11:
> openwrt CC 

OpenWrt Chaos Calmer 15.05 is unsupported.

closed 2020-03-11 by hnyman