openwrt/luci #3747: chrome open Overview page prompt error “Failed to execute 'open' on 'XMLHttpRequest': Invalid URL”

tty228, 2020-03-12:
Model: Phicomm K3
Architecture: ARMv7 Processor rev 0 (v7l)
Firmware Version: OpenWrt 19.07.2 / LuCI openwrt-19.07 branch git-20.068.59791-6b1a243
Kernel Version: 4.14.171
chrome Version: Chromium 78.0.3904.108

[image]

Internet Explorer Explorer nor Firefox shows this error.

I tried to troubleshoot. I found the reason why JavaScript asynchronous did not wait for a callback. After adding alert() or settimeout() to JavaScript code, it can run normally. After using the window.addeventlistener method, the number of errors decreased, but it still happens occasionally.

Is there any good way to help me?

Original code:
`<script type="text/javascript">L.require('view.status.index').catch(function(err) {`
`	L.dom.content(document.querySelector('#view'), null);`
`	L.error(err);`
`});</script>`

My current code:
`<script type="text/javascript">`
`function downloadJSAtOnload() {`
`	L.require('view.status.index').catch(function(err) {`
`	L.dom.content(document.querySelector('#view'), null);`
`	L.error(err);`
`});;`
`}`

`function load() {setTimeout(function(){downloadJSAtOnload();},200);}`

`	if (window.addEventListener)`
`		window.addEventListener("load",load, false); `
`	else if (window.attachEvent)`
`		window.attachEvent("onload",load);`
`	else `
`		window.onload = load;`
`</script>`

jsravn, 2020-03-26:
I also noticed this after updating to 19.07.2 from 19.07.2. Chrome can no longer open these pages.

tty228, 2020-03-29:
> I also noticed this after updating to 19.07.2 from 19.07.2. Chrome can no longer open these pages.

But I don't understand why this problem is rarely discussed. Is it the specific version of chrome that causes this problem.

cgi-bin/luci/admin/system/system    （system pages）
/cgi-bin/luci/admin/network/dhcp    （dhcp  pages）
sometimes have this problem. I can only add `setTimeout ()`,But it's too slow to open.

CanadianJeff, 2020-12-10:
why was this closed and moved to a chinese thread??? I still see this bug

n3gwg, 2021-12-04:
I am using Chrome Version 96.0.4664.55 (Official Build) (x86_64) on macOS Big Sur 11.6.1 and upon opening the main webpage of the router running OpenWRT I am experiencing the same issue.

jow-, 2021-12-04:
This issue has not been closed, a foreign chinese issue report referencing this one has been closed. This report is still open.

n3gwg, 2021-12-04:
I see it was the dupes not the actual issue that was closed.

commit 5663fd59 (2022-02-21): luci-base: properly handle promise targets in Request.request()

Under some circumstances, ubus RPC requests may be initiated while LuCI is
still resolving the `rpcBaseURL` value. In this situation, the `target`
argument of the `request()` call will be a pending promise object which
results in an invalid URL when serialized by `expandURL()`, leading to an
`Failed to execute 'open' on 'XMLHttpRequest': Invalid URL` exception.

This commonly occured on the index status page which immediately initiates
ubus RPC calls on load to discover existing status page partials.

Solve the issue by filtering [... cut]
commit 31a27f30 (2022-02-21): luci-base: properly handle promise targets in Request.request()

Under some circumstances, ubus RPC requests may be initiated while LuCI is
still resolving the `rpcBaseURL` value. In this situation, the `target`
argument of the `request()` call will be a pending promise object which
results in an invalid URL when serialized by `expandURL()`, leading to an
`Failed to execute 'open' on 'XMLHttpRequest': Invalid URL` exception.

This commonly occured on the index status page which immediately initiates
ubus RPC calls on load to discover existing status page partials.

Solve the issue by filtering [... cut]
commit 1b6c1262 (2022-02-21): luci-base: properly handle promise targets in Request.request()

Under some circumstances, ubus RPC requests may be initiated while LuCI is
still resolving the `rpcBaseURL` value. In this situation, the `target`
argument of the `request()` call will be a pending promise object which
results in an invalid URL when serialized by `expandURL()`, leading to an
`Failed to execute 'open' on 'XMLHttpRequest': Invalid URL` exception.

This commonly occured on the index status page which immediately initiates
ubus RPC calls on load to discover existing status page partials.

Solve the issue by filtering [... cut]
closed 2022-02-21 by jow-