openwrt/luci #3753: declarative menu.d/json support in 19.07?

dibdot, 2020-03-14:
@jow- 
currently I'm pre-testing adblock 4 here (https://forum.openwrt.org/t/adblock-4-pre-releases/57101). I've switched to client side javascript and declarative menu.d/json (without any lua controller at all).

In master this works quite fine, in 19.07 I didn't see any menu entries in LuCI, do I need a stub controller file or something like that?

jow-, 2020-03-14:
You need a dummy controller for 19.07 yet since the declarative menu JSON support is not backported yet.

closed 2020-03-14 by jow-

dibdot, 2020-03-15:
Thanks! I've added the controller and the app appeared in the menu of 19.07, too.
Now I'm running in rpc-/acl-related problems. I have the following json acl which is working fine in master:

```
{
	"luci-app-adblock": {
		"description": "Grant access to LuCI app adblock",
		"write": {
			"uci": ["adblock"],
			"file": {
				"/etc/adblock/*": ["read"],
				"/etc/crontabs/root": ["read"],
				"/var/run/adblock.pid": ["read"],
				"/tmp/adb_runtime.json": ["read"],
				"/etc/adblock/adblock.blacklist": ["write"],
				"/etc/adblock/adblock.whitelist": ["write"],
				"/sbin/logread -e adblock-": [ "exec" ],
				"/etc/init.d/adblock list" : [ "exec" ],
				"/etc/init.d/adblock start" : [ "exec" ],
				"/etc/init.d/adblock suspend" : [ "exec" ],
				"/etc/init.d/adblock resume" : [ "exec" ],
				"/etc/init.d/adblock timer * [0-9]* [0-9*]* [1-7,-*]*" : [ "exec" ],
				"/etc/init.d/adblock report * [0-9]* [a-z]* json" : [ "exec" ],
				"/etc/init.d/adblock query *" : [ "exec" ]
			}
		}
	}
}
```
While writing to the local blacklist is working fine, I get a "Permission denied" error whenever I'll try to exec e.g. the above logread command. Any hints are welcome! ;-)

dibdot, 2020-03-15:
I think I found the culprit: the "_direct" variants, e.g. "exec_direct" are not supported in 19.07 ...