openwrt/luci #3756: OpenWrt 19.07.2 broke two luci-mod-* files

ralfbergs, 2020-03-14:
Upgraded several systems from OpenWrt 19.07.0 or 1 to 19.07.2, and they all have partially broken Luci menus now (Firewall status and diagnostics). This is the offending lines from the installation:
```
Upgrading luci-mod-status on root from git-20.057.55219-13dd17f-1 to git-20.067.35699-cf18bdd-1...
Downloading http://downloads.openwrt.org/releases/19.07.2/packages/arm_cortex-a9_vfpv3/luci/luci-mod-status_git-20.067.35699-cf18bdd-1_arm_cortex-a9_vfpv3.ipk
Removing obsolete file /usr/lib/lua/luci/view/admin_status/dmesg.htm.
Removing obsolete file /usr/lib/lua/luci/view/admin_status/syslog.htm.
Removing obsolete file /usr/lib/lua/luci/view/admin_status/iptables.htm.
Removing obsolete file /usr/lib/lua/luci/view/admin_status/routes.htm.
Configuring luci-mod-status.
[...]
Upgrading luci-mod-network on root from git-20.057.55219-13dd17f-1 to git-20.067.35699-cf18bdd-1...
Downloading http://downloads.openwrt.org/releases/19.07.2/packages/arm_cortex-a9_vfpv3/luci/luci-mod-network_git-20.067.35699-cf18bdd-1_all.ipk
Removing obsolete file /usr/lib/lua/luci/view/admin_network/diagnostics.htm.
Configuring luci-mod-network.
```
Thanks.

ralfbergs, 2020-03-14:
Seems to be the root cause for #3736

jow-, 2020-03-14:
You need to upgrade *all* LuCI packages, clean the index cache or reboot.

closed 2020-03-14 by jow-

ralfbergs, 2020-03-14:
Thanks. First time I'm seeing these issues, but indeed after I mangled with the above, it was working again...