openwrt/luci #3757: Reporting security bugs on openwrt-19.07

alisaeed, 2020-03-15:
Hello, hope you are doing well,

We performed a security scan using acunetix and the results show some security issues.  Please see the attachment.
branch: openwrt-19.07
Vulnerability tool: Acunetix.Web.12.0.181218140.Retail

Best Regards.

[report.pdf](https://github.com/openwrt/luci/files/4334059/report.pdf)

closed 2020-03-15 by jow-

alisaeed, 2020-03-16:
Thanks a lot @jow- 
But yet, there are a few bugs, after your vulnerabilities fixing.
I attached a new report.

[20200316_Developer_https_192_168_1_100_cgi_bin_luci_.pdf](https://github.com/openwrt/luci/files/4337352/20200316_Developer_https_192_168_1_100_cgi_bin_luci_.pdf)

jow-, 2020-03-16:
Did you validate the claimed CSRF protection issue with a reliable reproducer? I am not going to fix the low and informational issues.

commit 5d7dc391 (2020-03-15): themes: ensure that data-page attribute is escaped

Fixes: #3757
commit 06884b95 (2020-03-15): themes: ensure that data-page attribute is escaped

Fixes: #3757

(cherry picked from commit 5d7dc391d4af4ad5dd6d7e1f6ef9891aa21fd220)