openwrt/luci #3758: "Associated Stations" are shown only for primary networks

luk1337, 2020-03-15:
It looks like Luci is trying to call `iwinfo radio1.network2 assoclist` rather than `iwinfo wlan1-1` thus receiving an empty response.

[image]

luk1337, 2020-03-15:
I'm assuming that this is cause only primary interfaces return `ifname` according to `ubus call network.wireless status` output:

```
root@OpenWrt:~# ubus call network.wireless status
{
	"radio0": {
		"up": true,
		"pending": false,
		"autostart": true,
		"disabled": false,
		"retry_setup_failed": false,
		"config": {
			"hwmode": "11g",
			"path": "pci0000:00/0000:00:00.0/0000:01:00.0",
			"noscan": "1",
			"noscan": true,
			"htmode": "HT40",
			"channel": "auto"
		},
		"interfaces": [
			{
				"section": "default_radio0",
				"ifname": "wlan0",
				"config": {
					"mode": "ap",
					"key": "-snip-",
					"ssid": "-snip-",
					"encryption": "psk2",
					"network": [
						"lan"
					],
					"mode": "ap"
				}
			}
		]
	},
	"radio1": {
		"up": true,
		"pending": false,
		"autostart": true,
		"disabled": false,
		"retry_setup_failed": false,
		"config": {
			"hwmode": "11a",
			"path": "pci0000:00/0000:00:01.0/0000:02:00.0",
			"htmode": "VHT80",
			"channel": "auto"
		},
		"interfaces": [
			{
				"section": "default_radio1",
				"ifname": "wlan1",
				"config": {
					"mode": "ap",
					"key": "-snip-",
					"ssid": "-snip-",
					"encryption": "psk2",
					"network": [
						"lan"
					],
					"mode": "ap"
				}
			},
			{
				"section": "wifinet2",
				"config": {
					"ssid": "-snip-",
					"encryption": "sae",
					"mode": "ap",
					"ieee80211w": 2,
					"key": "-snip-",
					"network": [
						"lan"
					],
					"mode": "ap"
				}
			}
		]
	}
}
```

jow-, 2020-03-15:
Thats a bug in OpenWrt / netifd, not in LuCI. Without the ifname information, it is impossible to reliably map the wireless interface information.

jow-, 2020-03-20:
Closing this here as it needs to be fixed on the OpenWrt side.

closed 2020-03-20 by jow-