openwrt/luci #3759: luci-theme-rosy: Fonts not loaded, 19.07.2

ngerke, 2020-03-16:
```
Model	Netgear R6250 V1 (BCM4708)
Architecture	ARMv7 Processor rev 0 (v7l)
Firmware Version	OpenWrt 19.07.2 r10947-65030d81f3 / LuCI openwrt-19.07 branch git-20.067.35699-cf18bdd

luci-theme-rosy	git-20.067.35699-cf18bdd-1	~105.1 KB	Rosy Theme
```

The theme fonts is not loaded:

[image]

jow-, 2020-03-16:
The theme is marked broken since a while and I'll likely drop it from the tree soon if nobody steps up to maintain it.

ngerke, 2020-03-16:
Oh thats to bad, I'm new to OpenWRT where would I've found the 'broken pkg' notice?

jow-, 2020-03-16:
The flag wasn't applied to the OpenWrt 19.07 release, I did that now so the package won't be built anymore with the next release.

ysc3839, 2020-03-16:
Ping @rosysong

HashedBrown, 2020-03-17:
bummer. it's a nice look and feel. Just the font is too huge.

hnyman, 2020-12-29:
The broken theme has been removed.

closed 2020-12-29 by hnyman