openwrt/luci #3760: luci-app-statistics: no IPv6 firewall view (19.07.1)

oofnikj, 2020-03-16:
I added some monitoring for packet/byte counters both for ipv4 and ipv6 using luci-app-statistics / collectd. After verifying that the data was being collected properly, I didn't see any way to view the data in the statistics page on LuCI for IPv6.

So I created a rrdtool definitions file in `/usr/lib/lua/luci/statistics/rrdtool/definitions/ip6tables.lua` ([contents](https://github.com/oofnikj/luci/blob/openwrt-19.07/applications/luci-app-statistics/luasrc/statistics/rrdtool/definitions/ip6tables.lua)) and now my ip6tables rules are rendering graphs as well.

Can I submit a PR for this or is it no longer relevant since it seems LuCI is moving to client-side rendering?

jow-, 2020-03-16:
Yes, please submit a PR. I am not yet sure if the client side LuCI stats will be ported to 19.07

jow-, 2020-03-21:
Merged in 4778aa6 - thanks!

closed 2020-03-21 by jow-