openwrt/luci #3765: Incomplete configuration when zone name equals interface name

Zepmann, 2020-03-18:
## Summary

When the user uses LUCI to create a new firewall zone for an interface and gives them both the same name, an incomplete uCI firewall configuration is generated. This is dangerous, since the new firewall zone will not apply to the interface.

## OpenWRT information

Tested on OpenWRT 19.07.2 (current stable).

## Reproduce the issue

1. Create a new interface. For this example 'testinterface' is used as its name.
1. On the Network -> Firewall page, create a new firewall zone and assign it to the interface. Give it a **different** name. For this example 'testzone' is used.

The resulting uCI commands are as follows:
```
uci add firewall zone # =cfg11dc81
uci set firewall.@zone[-1].network='testzone testinterface'
uci set firewall.@zone[-1].name='testzone'
uci set firewall.@zone[-1].mtu_fix='1'
uci set firewall.@zone[-1].input='REJECT'
uci set firewall.@zone[-1].forward='REJECT'
uci set firewall.@zone[-1].masq='1'
uci set firewall.@zone[-1].output='ACCEPT'
```

Please pay attention to the line stating `uci set firewall.@zone[-1].network='testzone testinterface'`. This is important.

Now remove the zone and the interface, and restart the experiment:

1. Create a new interface. For this example 'test' is used as its name.
2. On the Network -> Firewall page, create a new firewall zone and assign it to the interface. Also name it 'test'.

The resulting uCI commands are:
```
uci add firewall zone # =cfg11dc81
uci set firewall.@zone[-1].name='test'
uci set firewall.@zone[-1].mtu_fix='1'
uci set firewall.@zone[-1].input='REJECT'
uci set firewall.@zone[-1].forward='REJECT'
uci set firewall.@zone[-1].masq='1'
uci set firewall.@zone[-1].output='ACCEPT'
```

As you can see, a line with the .network parameter is missing. The result of this is that the newly created zone does not cover the interface interface as it was configured in the GUI.

## Actual behavior

An incomplete firewall zone is created since the following line is not generated:
```
uci set firewall.@zone[-1].network='test test'
```

## Expected behavior

The following line should be part of the generated configuration to be consistent with the 'testzone testinterface' example:
```
uci set firewall.@zone[-1].network='test test'
```

Furthermore, it is unclear to me why the name of the firewall zone is part of the .network parameter, as shown in the first example of 'testzone testinterface'. Would 'testinterface' simply not be enough? If that is true, `uci set firewall.@zone[-1].network='test'` would also be enough.

jow-, 2020-03-18:
This has already been fixed with 4052436d8 and b01cf6953.

closed 2020-03-18 by jow-

Zepmann, 2020-03-18:
Thanks for the clarification, @jow- . Good to see that it is fixed.