openwrt/luci #3768: luci-app-dcwapd: Installation error

moocan, 2020-03-20:
Hello,

I hope that's the right place to do it.
After installing on OpenWRT 19.07.2 (device: NetGear R7800) `luci-app-dcwapd` with standard `opkg install luci-app-dcwapd` luci seems to be corrupted.

Here is the error
```
/usr/lib/lua/luci/dispatcher.lua:655: Invalid controller file found
The file '/usr/lib/lua/luci/controller/dcwapd.lua' contains an invalid module line.
Please verify whether the module name is set to 'luci.controller.dcwapd' - It must correspond to the file path!
stack traceback:
	[C]: in function 'assert'
	/usr/lib/lua/luci/dispatcher.lua:655: in function 'createindex'
	/usr/lib/lua/luci/dispatcher.lua:681: in function 'createtree'
	/usr/lib/lua/luci/dispatcher.lua:347: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:208: in function </usr/lib/lua/luci/dispatcher.lua:207>
```

Thank you in adavncefor your advices.

Kind Regards

jow-, 2020-03-20:
@csonsino

csonsino, 2020-03-23:
I just performed a clean install of 19.07.2 (prebuilt upgrade binary from https://openwrt.org/toh) on a WNDR3700, and here's what I got:
```
root@OpenWrt:~# opkg update
Downloading http://downloads.openwrt.org/releases/19.07.2/targets/ath79/generic/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_core
Downloading http://downloads.openwrt.org/releases/19.07.2/targets/ath79/generic/packages/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/releases/19.07.2/targets/ath79/generic/kmods/4.14.171-1-b84a5a29b1d5ae1dc33ccf9ba292ca1d/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_kmods
Downloading http://downloads.openwrt.org/releases/19.07.2/targets/ath79/generic/kmods/4.14.171-1-b84a5a29b1d5ae1dc33ccf9ba292ca1d/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/releases/19.07.2/packages/mips_24kc/base/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_base
Downloading http://downloads.openwrt.org/releases/19.07.2/packages/mips_24kc/base/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/releases/19.07.2/packages/mips_24kc/luci/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_luci
Downloading http://downloads.openwrt.org/releases/19.07.2/packages/mips_24kc/luci/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/releases/19.07.2/packages/mips_24kc/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_packages
Downloading http://downloads.openwrt.org/releases/19.07.2/packages/mips_24kc/packages/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/releases/19.07.2/packages/mips_24kc/routing/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_routing
Downloading http://downloads.openwrt.org/releases/19.07.2/packages/mips_24kc/routing/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/releases/19.07.2/packages/mips_24kc/telephony/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_telephony
Downloading http://downloads.openwrt.org/releases/19.07.2/packages/mips_24kc/telephony/Packages.sig
Signature check passed.
root@OpenWrt:~# opkg install luci-app-dcwapd
Installing luci-app-dcwapd (git-20.081.71863-12f0dac-1) to root...
Downloading http://downloads.openwrt.org/releases/19.07.2/packages/mips_24kc/luci/luci-app-dcwapd_git-20.081.71863-12f0dac-1_all.ipk
Installing luci-compat (git-20.081.71863-12f0dac-1) to root...
Downloading http://downloads.openwrt.org/releases/19.07.2/packages/mips_24kc/luci/luci-compat_git-20.081.71863-12f0dac-1_all.ipk
Installing uclibcxx (0.2.5-3) to root...
Downloading http://downloads.openwrt.org/releases/19.07.2/packages/mips_24kc/base/uclibcxx_0.2.5-3_mips_24kc.ipk
Installing kmod-br-netfilter (4.14.171-1) to root...
Downloading http://downloads.openwrt.org/releases/19.07.2/targets/ath79/generic/kmods/4.14.171-1-b84a5a29b1d5ae1dc33ccf9ba292ca1d/kmod-br-netfilter_4.14.171-1_mips_24kc.ipk
Installing kmod-macremapper (4.14.171-ath79-1) to root...
Downloading http://downloads.openwrt.org/releases/19.07.2/targets/ath79/generic/kmods/4.14.171-1-b84a5a29b1d5ae1dc33ccf9ba292ca1d/kmod-macremapper_4.14.171-ath79-1_mips_24kc.ipk
Installing libdcwsocket (1.1.0-1) to root...
Downloading http://downloads.openwrt.org/releases/19.07.2/packages/mips_24kc/packages/libdcwsocket_1.1.0-1_mips_24kc.ipk
Installing libdcwproto (1.1.0-1) to root...
Downloading http://downloads.openwrt.org/releases/19.07.2/packages/mips_24kc/packages/libdcwproto_1.1.0-1_mips_24kc.ipk
Installing mrmctl (1.1.0-1) to root...
Downloading http://downloads.openwrt.org/releases/19.07.2/packages/mips_24kc/packages/mrmctl_1.1.0-1_mips_24kc.ipk
Installing dcwapd (1.1.0-2) to root...
Downloading http://downloads.openwrt.org/releases/19.07.2/packages/mips_24kc/packages/dcwapd_1.1.0-2_mips_24kc.ipk
Configuring luci-compat.
Configuring uclibcxx.
Configuring kmod-br-netfilter.
Configuring kmod-macremapper.
Configuring libdcwsocket.
Configuring libdcwproto.
Configuring mrmctl.
Configuring dcwapd.
dcwapd is disabled in UCI
Configuring luci-app-dcwapd.
root@OpenWrt:~#
```

Did your opkg install attempt to install all of the dependency packages?

Maybe try to remove it and reinstall? (not sure if you'd need to add `--force-remove` or `--force-removal-of-dependent-packages`)

moocan, 2020-03-25:
Hello,

Many thanks for your reply.
I don't understand.
I removed using `--force-removal-of-dependent-packages` and then reinstalled and after with just `--force-remove` 4 times and now no errors in lucy. I can access the administration page and dcwapd page. I don't have tested yet.
I don't know if it could be linked but I chose to use lucy with nginx.
I don't have any extra packages just an extroot config.

Thank you for your support.
Kind Regards

csonsino, 2020-03-26:
Did you build your image, or download a prebuilt?  If you built it, I suppose that the combination of packages that you've selected/installed could be causing something strange.  If you want to paste a diffconfig (in your openwrt directory, do `./scripts/diffconfig.sh > moocan.diffconfig`), I can try to build something as close to what you have.

feckert, 2022-04-12:
No Comments since two years anymore so I assume this is fixed

closed 2022-04-12 by feckert