openwrt/luci #3773: Error when opening luci with latest 18.06.8

svollebregt, 2020-03-20:
I am running OpenWRT on a TP-Link WA850REv1 which has ar71xx-ath79 as target. After performing a sysupgrade using the build available on openwrt I get the following error:
`
/usr/lib/lua/luci/dispatcher.lua:234: /etc/config/luci seems to be corrupt, unable to find section 'main'
stack traceback:
	[C]: in function 'assert'
	/usr/lib/lua/luci/dispatcher.lua:234: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:127: in function </usr/lib/lua/luci/dispatcher.lua:126>
`

This is the content of my luci config:
`
config core 'main'
        option lang 'auto'
        option mediaurlbase '/luci-static/bootstrap'
        option resourcebase '/luci-static/resources'

config extern 'flash_keep'
        option uci '/etc/config/'
        option dropbear '/etc/dropbear/'
        option openvpn '/etc/openvpn/'
        option passwd '/etc/passwd'
        option opkg '/etc/opkg.conf'
        option firewall '/etc/firewall.user'
        option uploads '/lib/uci/upload/'

config internal 'languages'

config internal 'sauth'
        option sessionpath '/tmp/luci-sessions'
        option sessiontime '3600'

config internal 'ccache'
        option enable '1'

config internal 'themes'
        option Bootstrap '/luci-static/bootstrap'

config internal 'apply'
        option rollback '30'
        option holdoff '4'
        option timeout '5'
        option display '1.5'

config internal 'diag'
        option dns 'openwrt.org'
        option ping 'openwrt.org'
        option route 'openwrt.org'
`

Reverting back to 18.06.7 the issue disappears.

jow-, 2020-03-20:
No, it is still a mystery what is causing the issue exactly. So far it seems to be caused by some backed up configuration but neither `/etc/config/luci` nor `/etc/config/rpcd` seem to be related.

Would you be able to send me the contents of your `/overlay` via mail. You might need to redact passwords and the like.

svollebregt, 2020-03-21:
Thank you for your reply, I just send you the content of my /overlay

JohnHoder, 2020-04-10:
I am facing the same problem - device ASUS RT-AC65P.
I tried to play around with it a little bit and what seemed odd to me was:
- require function which takes config file as argument does not succeed to find the config file at all it seems - but, on the other side, when I rewrite require "luci.config" to require "/etc/config/luci" that passes but crashed a few lines further.

I faced problems at line 314 and 234 in dispatcher.lua

fire-burner, 2020-05-25:
I had the same issue on my TL-WA730RE v1 after upgrading from 18.06.7 to 18.06.8.

searching the web brought me to a forum post: https://forum.openwrt.org/t/recent-upgrade-broke-luci/57427/48
alioth advised to add
```
config rpcd
option socket /var/run/ubus.sock
option timeout 30
```
to /etc/config/rpcd

After adding this part to my /etc/config/rpcd and restarting, luci is reachable normally as before.

svollebregt, 2020-05-30:
I can confirm that the addition suggested by fire-burner resolves the issue for me.

zorun, 2020-06-13:
FYI, I traced it back to a libubox regression: https://bugs.openwrt.org/index.php?do=details&task_id=3177

jow-, 2020-07-04:
Closing this here as it is an OpenWrt issue.

closed 2020-07-04 by jow-