openwrt/luci #3782: luci-app-upnp Not always opening ports for Destiny 2 PC

azure-xiii, 2020-03-25:
Basically whenever I start up Destiny 2 on PC I have a about a 1 in 10 chance of UPNP (using luci-app-upnp) opening the required ports. I have IGDv1 enabled. This issue happens only on Openwrt (every version I have tried)

If I use the stock Netgear firmware or my ASUS router with stock firmware... it will open the port everytime. I know I can manually port forward but I have two people who play Destiny in the house and would like to use UPNP if possible. I really prefer using Openwrt due to CAKE

Device - R7800
OpenWrt Versions - Generic 19.07.2, Generic OpenWrt 18.06.8, Hynmans last 18.06, multiple Hynmans Master, and multiple Hynmans 19.07 releases.

Thanks!

solidus1983, 2020-03-29:
Have you setup the device playing D2 in question in the upnp section?

Also would help for a syslog dump with sensitive info removed.

Make sure upnp is in verbose mode reboot the router/modem then run D2 if it doesn't show open NAT or upnp port then grab the system log remove the sensitive info and attach it.

jow-, 2020-05-07:
Closing this as LuCI merely displays the upnp status. Miniupnpd bugs should be reported at https://github.com/openwrt/packages

closed 2020-05-07 by jow-