openwrt/luci #3795: LuCI dynlist text value overflow when long input value is present

jamesmacwhite, 2020-03-29:
A minor visual bug, demonstrated with dnsmasq, when a long value is present in a dynlist, the text can overflow and breakout of the input area.

[image].

A simple fix can be done with CSS targeting the rule `.cbi-dynlist > .item` and applying `overflow: hidden;`, preventing the value in the `<span>` from going outside of the div.

jow-, 2020-04-03:
Thank your for the suggestion. I added a word wrap in addition to the overflow so that information is not "lost". Looks like this for long strings now:

[image]

closed 2020-04-03 by jow-

jamesmacwhite, 2020-04-03:
@jow- Thanks for looking at this. Good idea, even better with also adding `work-break` , so you can see the whole value in it's entirety!

commit 015712bb (2020-04-03): luci-theme-bootstrap, luci-theme-openwrt-2020: fix dynlist overflows

Fixes: #3795