openwrt/luci #3800: map-e cannot save in luci and map-e cannot work in mwan3

tinysun, 2020-03-31:
map-e cannot save in luci ,sameone else have the same issuse in 19.07.02
 map-e cannot work in mwan3,when I use 2wan with map-e and pppoe.at first it looks like fine.but after a few min,the map-e line is down.

jow-, 2020-04-10:
What means "cannot save"?

tinysun, 2020-04-11:
In luci I can edit the page of map-e, but when I want to save the setting is not finish in 19.07.02

feckert, 2020-04-14:
Your description of the error is very vague! What does map-e mean here? Please give us more information so we can help you.

tinysun, 2020-04-14:
In luci ，I must to create a new wan to use map-e。when I  create a new wan with map-e protocol，I can Click the save button，but I cannot finish。

feckert, 2020-04-17:
Eexcuse me your remarks and problem are not entirely clear.
Please ask in the openwrt [forum](https://forum.openwrt.org/)
Maybe some one else could help you there.

closed 2020-04-17 by feckert