openwrt/luci #3803: Modal windows in openwrt2020 theme

dibdot, 2020-03-31:
@jow Just a quick one: is this a theme issue or is this an adblock issue (see screenshots)? In the older themes it looks OK, so I ask ...
[image]
[image]

jow-, 2020-03-31:
This happens because the modal dialog in OpenWrt2020 is wider than the old themes. If you resize your browser window to become a little narrower you should see the elements snapping in place.

To make the layout more rigid, I suggest removing the `float: left;` directives and adding `style="display:flex; flex-direction:column"` to the parent `<div class="left">` instead.

dibdot, 2020-03-31:
Indeed, you're right - thank you!

closed 2020-03-31 by dibdot