openwrt/luci #3818: baseclass.js seems to be missing from the repo

DHowett, 2020-04-03:
https://github.com/openwrt/luci/commit/3c4bc228a1f7b5731cf464f3f407c9ed9ace3cd2 introduced a dependency on `baseclass`, which cannot be loaded and doesn't ive in luci-base/htdocs.

```
HTTP error 404 while loading class file "/luci-static/resources/baseclass.js?v=git-19.292.31773-cc35206"
  at compileClass (https://arcturus.delfino.n.howett.net/luci-static/resources/luci.js?v=git-19.292.31773-cc35206:110:3)
  at promise callback*require (https://arcturus.delfino.n.howett.net/luci-static/resources/luci.js?v=git-19.292.31773-cc35206:122:112)
  at compileClass (https://arcturus.delfino.n.howett.net/luci-static/resources/luci.js?v=git-19.292.31773-cc35206:112:151)
  at promise callback*require (https://arcturus.delfino.n.howett.net/luci-static/resources/luci.js?v=git-19.292.31773-cc35206:122:112)
  at compileClass (https://arcturus.delfino.n.howett.net/luci-static/resources/luci.js?v=git-19.292.31773-cc35206:112:151)
```

```
root@arcturus:~# ls /www/luci-static/resources/
bandwidth.svg    firewall.js      load.svg         protocol         ui.js            wireless.svg
cbi              form.js          luci.js          rpc.js           validation.js    xhr.js
cbi.js           fs.js            network.js       tools            view
connections.svg  icons            promis.min.js    uci.js           wifirate.svg
```

jow-, 2020-04-03:
You're missing 711f75927849fade74f79e4b198b3a839d9d4fbc

closed 2020-04-03 by jow-

DHowett, 2020-04-03:
Er, I checked the repository _as of master_ and it doesn't exist. I don't believe I can have https://github.com/openwrt/luci/commit/3c4bc228a1f7b5731cf464f3f407c9ed9ace3cd2 without https://github.com/openwrt/luci/commit/711f75927849fade74f79e4b198b3a839d9d4fbc (since git is a DAG...)

jow-, 2020-04-03:
Maybe I was a bit terse in my reply. Judging from the stack trace you posted, you're running revision cc35206 which is not an official LuCI commit. Also the timestamp `19.292.31773` indicates a rather old version from some time last year. Did you maybe install new LuCI components on top of an old `luci-base`?

The error your describe can only happen if `luci-base`, or more specifically `luci.js` wasn't updated.

DHowett, 2020-04-03:
Ugh, thanks. There's a couple things here, it turns out. You pointing out an outdated/unofficial commit made me check everything out, and the results were at least a bit puzzling.

* I can't find `cc3520` in my feeds/ repos or reflogs either, which is puzzling
* Invalidating my browser cache, predictably, made it go away
* After that, I'm sitting at LuCI Master git-20.093.71480-89a50c2 _exactly as expected_

Sorry to raise the alarm so quickly and so curtly. User error is, as always, a real bear. :smile: