openwrt/luci #3835: luci-app-nlbwmon: nlbw.js?v=git-20.095.26180-bee97d8:16 Uncaught (in promise) ReferenceError: dom is not defined

peterwillcn, 2020-04-04:
luci-app-nlbwmon
```
nlbw.js?v=git-20.095.26180-bee97d8:16 Uncaught (in promise) ReferenceError: dom is not defined
    at pie (nlbw.js?v=git-20.095.26180-bee97d8:16)
    at renderHostData (nlbw.js?v=git-20.095.26180-bee97d8:67)
    at nlbw.js?v=git-20.095.26180-bee97d8:41
    at luci.js?v=git-19.292.31773-cc35206:203
```

jow-, 2020-04-04:
You need to upgrade `luci-base` and flush your caches.

peterwillcn, 2020-04-04:
The latest version
```
Package: luci-base
Version: git-20.095.26180-bee97d8-1
Depends: libc, lua, luci-lib-nixio, luci-lib-ip, rpcd, libubus-lua, luci-lib-jsonc, liblucihttp-lua, rpcd-mod-file, rpcd-mod-luci, cgi-io
Status: install ok installed
Section: luci
Architecture: x86_64
Size: 145364
Filename: luci-base_git-20.095.26180-bee97d8-1_x86_64.ipk
Conffiles:
 /etc/luci-uploads/.placeholder e3b0c44298fc1c149afbf4c8996fb92427ae41e4649b934ca495991b7852b855
 /etc/config/luci 60078f7c02ed91d23092c9b75cb9ea5c8ee39f7727def813c8201f8e48a296f4
 /etc/config/ucitrack 3480e8ca8e224537c9d95e7678b05123ff7b06a2a0b3d1000032a04576d90381
Description: LuCI core libraries
Installed-Time: 1585992756
```

jow-, 2020-04-04:
You are right, this file got accidentally converted by 3c4bc228a1f7b5731cf464f3f407c9ed9ace3cd2. I reverted it for now.

closed 2020-04-04 by jow-

peterwillcn, 2020-04-04:
@jow- Thank you

commit af6fd6df (2020-04-04): luci-app-nlbwmon: use legacy `L.dom` attribute

Until it is converted to a proper view, we need to use the legacy alias.

Fixes: #3835