openwrt/luci #3836: luci-app-dawn 500 Internal Server Error

tapper82, 2020-04-04:
Hi when I click on Network Overview I get.

500 Internal Server Error
Sorry, the server encountered an unexpected error.
/usr/lib/lua/luci/template.lua:97: Failed to execute template 'cbi/map'.
A runtime error occurred: /usr/lib/lua/luci/model/cbi/dawn/dawn_network.lua:8: module 'luci.json' not found:
	no field package.preload['luci.json']
	no file './luci/json.lua'
	no file '/usr/share/lua/luci/json.lua'
	no file '/usr/share/lua/luci/json/init.lua'
	no file '/usr/lib/lua/luci/json.lua'
	no file '/usr/lib/lua/luci/json/init.lua'
	no file './luci/json.so'
	no file '/usr/lib/lua/luci/json.so'
	no file '/usr/lib/lua/loadall.so'
	no file './luci.so'
	no file '/usr/lib/lua/luci.so'
	no file '/usr/lib/lua/loadall.so'
stack traceback:
	[C]: in function 'require'
	/usr/lib/lua/luci/model/cbi/dawn/dawn_network.lua:8: in function 'render'
	/usr/lib/lua/luci/cbi.lua:266: in function 'render_children'
	[string "/usr/lib/lua/luci/view/cbi/map.htm"]:6: in main chunk
stack traceback:
	[C]: in function 'error'
	/usr/lib/lua/luci/template.lua:97: in function 
	(tail call): ?
	/usr/lib/lua/luci/cbi.lua:257: in function 'render'
	/usr/lib/lua/luci/cbi.lua:440: in function 'render'
	/usr/lib/lua/luci/dispatcher.lua:1325: in function '_cbi'
	/usr/lib/lua/luci/dispatcher.lua:927: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:441: in function 
Powered by LuCI Master (git-20.095.36093-af6fd6d) / OpenWrt SNAPSHOT r12833-02640f0147

hnyman, 2020-04-10:
cc @PolynomialDivision

PolynomialDivision, 2020-04-10:
Hmmm, the dependencies are missing:
Workaround:

    opkg install luci-compat luci-lib-json

Weird, I thought that they have to be removed when I tried to bring the app upstream.
I will fix this, soon in upstream.

hnyman, 2020-04-10:
> thought that they have to be removed when I tried to bring the app upstream

Just a comment there:  I complained in your original PR, as you had the meta-package "luci" as dependency. That was wrong.

closed 2020-04-10 by hnyman

PolynomialDivision, 2020-04-10:
Thanks for clarification. :)

tapper82, 2020-04-10:
@tapper82 PolynomialDivision  Thanks for quick fix. ☺

PR #3868 (2020-04-10): luci-app-dawn: fix typos and dependencies
Fixes #3836.
Needs https://github.com/openwrt/packages/pull/11815.
commit 0ea4115a (2020-04-10): luci-app-dawn: add dependencies

Missing:
- luci-compat
- luci-lib-json

Fixes #3836