openwrt/luci #3837: menu is visible without authtentication

lucize, 2020-04-04:
using nginx I have this

[image]

lucize, 2020-04-04:
also
```
Sat Apr  4 17:49:46 2020 daemon.err nginx[23101]: 2020/04/04 17:49:46 [error] 23105#0: *19 open() "/www/luci-static/resources/protocol/gre.js" failed (2: No such file or directory), client: 1.2.3.4, server: _lan, request: "GET /luci-static/resources/protocol/gre.js?v=git-19.292.31773-cc35206 HTTP/1.1", host: "openwrt.domain", referrer: "http://openwrt.domain/cgi-bin/luci/admin/status/overview"
Sat Apr  4 17:49:46 2020 daemon.err nginx[23101]: 2020/04/04 17:49:46 [error] 23105#0: *22 open() "/www/luci-static/resources/protocol/gretap.js" failed (2: No such file or directory), client: 1.2.3.4, server: _lan, request: "GET /luci-static/resources/protocol/gretap.js?v=git-19.292.31773-cc35206 HTTP/1.1", host: "openwrt.domain", referrer: "http://openwrt.domain/cgi-bin/luci/admin/status/overview"
Sat Apr  4 17:49:46 2020 daemon.err nginx[23101]: 2020/04/04 17:49:46 [error] 23105#0: *24 open() "/www/luci-static/resources/protocol/relay.js" failed (2: No such file or directory), client: 1.2.3.4, server: _lan, request: "GET /luci-static/resources/protocol/relay.js?v=git-19.292.31773-cc35206 HTTP/1.1", host: "openwrt.domain", referrer: "http://openwrt.domain/cgi-bin/luci/admin/status/overview"

```

hnyman, 2020-04-04:
Regarding "menu is visible" part of your issue: 
Duplicate of #3653, #3766, #3785 and original discussion in #3563

closed 2020-04-04 by lucize