openwrt/luci #3840: Can't delete firewall zones on Archer C7 AC1750

anderbytes, 2020-04-05:
I simply cannot delete any firewall zones on my router. When clicked, the "DELETE" button does absolutely nothing, not even a popup message of error, unsaved change or anything.

In Firefox, console shows the error: "**ReferenceError: zone is not defined**"
In Chrome, the error is : "**Uncaught (in promise) ReferenceError: zone is not defined
    at eval (firewall.js?v=git-20.057.55219-13dd17f:24)**"

I found myself having to manually edit /etc/config/firewall and removing the zones for it to work properly.

Thanks.

jow-, 2020-04-05:
This looks like an error that has been fixed with 4c55e8226d7d6808ad7aa7d9a420dfda628a89f0. Your `luci-base` version seems to be older than that.

closed 2020-04-05 by jow-

anderbytes, 2020-04-05:
Hope this helps! Will upgrade soon and test again! Thanks