openwrt/luci #3843: Luci starts openvpn only if enable box is ticked on TP-Link Archer C50 HW v3

dolhastelian, 2020-04-05:
luci	git-20.094.46635-de52000-1
openvpn-openssl	2.4.7-2
luci-app-openvpn	git-20.094.46635-de52000-1
Firmware Version	OpenWrt 19.07.2 r10947-65030d81f3 / LuCI openwrt-19.07 branch git-20.094.46635-de52000
Kernel Version	4.14.171

From command line, using ssh, openvpn can be started / stopped without enabling it at startup

From Luci, start works only if enable box is ticked

audetto, 2020-04-26:
I had the same issue with Open WRT on a raspberry pi.

Powered by LuCI openwrt-19.07 branch (git-20.057.55219-13dd17f) / OpenWrt 19.07.2 r10947-65030d81f3 

I am referring to open vpn client connecting to 3rd party servers.
Just pressing start it does not do anything, one has to enable it to run disable to stop.

The missing use case is to quickly stop a client and start another one without affecting which one is enabled at startup.

adelton, 2025-01-10:
Hello @sbyx, as the original author, do you have any guidance about the interaction of the enabled vs. start/stop? Reading the code I truggle a bit to figure out where/how that enabled flag is actually set and used ...