openwrt/luci #3845: LuCI stuck at "Loading view..."

locojohn, 2020-04-05:
LuCI interface is stuck and a message "Loading view..." appears at any page I navigate and then hangs indefinitely.

Using latest snapshot: LuCI Master (git-20.096.39557-e7d935d) / OpenWrt SNAPSHOT r12839-95bd6a04b5
Kernel: 5.4.28

/etc/config/uhttpd:

config uhttpd 'main'
        list listen_https '0.0.0.0:443'
        option home '/www'
        option rfc1918_filter '0'
        option max_requests '3'
        option max_connections '5'
        option cgi_prefix '/cgi-bin'
        option script_timeout '60'
        option network_timeout '30'
        option http_keepalive '20'
        option tcp_keepalive '0'
        option ubus_prefix '/ubus'
        option realm 'MyRouter Home Network'
        option key '/root/.acme.sh/home.myrouter.org/home.myrouter.org.key'
        option cert '/root/.acme.sh/home.myrouter.org/fullchain.cer'

config cert 'px5g'
        option days '730'
        option bits '2048'
        option country 'LV'
        option state 'Riga'
        option commonname 'home.myrouter.org'

jow-, 2020-04-05:
Press `F12` and look for errors.

locojohn, 2020-04-05:
I am using Safari.  Neverheless, the issue went away after I rebuilt and re-installed the new firmware on my router.  Perhaps, it was a temporary glitch.

closed 2020-04-05 by locojohn