openwrt/luci #3860: luci-app-sqm: bad run_path variable initialization

dl12345, 2020-04-08:
On the majority of Openwrt systems, `/var` is a symlink to `/tmp`

However, if the router has a SSD or writeable storage of some form and `/var` is made to be a normal directory in the filesystem instead of a symlink to `/tmp`, the initialization of the run_path variable in sqm-cbi.lua causes breakage of sqm-scripts functionality in the sense thaty only fq_codel is shown as an available qdisc.

run_path is explicitly initialized in sqm-cbi.lua to `/tmp/run/sqm/available_qdiscs` instead of the more correct and portable `/var/run/sqm/available_qdiscs`

```
local wa = require "luci.tools.webadmin"                                             
local fs = require "nixio.fs"                                                    
local net = require "luci.model.network".init()              
local sys = require "luci.sys"                                                                                 
--local ifaces = net:get_interfaces()                          
local ifaces = sys.net:devices()                                  
local path = "/usr/lib/sqm"                                   
local run_path = "/tmp/run/sqm/available_qdiscs"
```

hnyman, 2020-04-08:
cc @tohojo @moeller0

tohojo, 2020-04-08:
Yeah, that seems like an odd oversight; let me fix that...

commit 88ec332e (2020-04-08): luci-app-sqm: Fix run path

luci-app-sqm was using the wrong path to look for the available_qdiscs
file. This happened to work because by default /var is symlinked to /tmp on
OpenWrt. However, it breaks badly when this is not the case, and the
sqm.conf file for OpenWrt uses /var/run anyway, so update the luci code to
do this as well.

Fixes #3860.
closed 2020-04-08 by tohojo