openwrt/luci #3864: luci-app-wol sends a wrong etherwake command.

Paolo177, 2020-04-09:
The problem is very simple but caused me many hours of work to figure it out:

luci-app-wol is just the GUI for etherwake commands BUT it sends commands with MAC addresses in capital letters 1A:2B:3C:D4:E5:F6 but my pc wakes up only if it receives the command with the MAC in lowercase 1a:2b:3c:d4:e5:f6 so the solution is pretty simple, make luci-app-wol send the command with the lowercase command.

ysc3839, 2020-04-10:
I think you don't know how wake on LAN works. Wake on LAN using a protocol called magic packet. The MAC address in magic packet is raw bytes, instead of ASCII text. So there's no upper or lower case.
I think it's a bug of etherwake. You can use [Wireshark](https://www.wireshark.org/) to check if the packet is actually sent.

Paolo177, 2020-04-10:
@ysc3839 I totally agree with you, that's crazy. You don't know ho many hours of work I spent to resolve the issue. I used wireshark, I used the tcpdump command on my freenas to see if packet arrived, i used so many tools to try to understand why [this windows program](https://www.depicus.com/wake-on-lan/wake-on-lan-gui) was ALWAYS able to wake my freenas pc but luci-app-wol isn't.

I of course went in ssh and wrote so many commands like:
/usr/bin/etherwake -D -b "1A:2B:3C:D4:E5:F6"
/usr/bin/etherwake -D -i "br-lan" -b "1A:2B:3C:D4:E5:F6"
/usr/bin/etherwake -D -i br-lan -b 1A:2B:3C:D4:E5:F6
/usr/bin/etherwake 1A:2B:3C:D4:E5:F6 -D
and at least 100 other that don't make any sense, i was able to see the packet with wireshark, they arrived to my freenas, seen with tcpdump, but they never woke it up!

Untill I used this command:

/usr/bin/etherwake -i br-lan 1a:2b:3c:d4:e5:f6

this command up here just works, any other command does not.

Please believe me and save the same amount of time spent figuring it out to the next guy :)

closed 2020-04-10 by Paolo177