openwrt/luci #3867: untranslatable strings / grammar constraints

ce-4, 2020-04-10:
Hi openwrt devs,

Several strings are build using concatenation following english grammar / word order rules which is incompatible with eg. German word order:

E.g.
`errorTable["errorRestoreCache"] = translate("failed to move") .. " '" .. outputCache .. "' " .. translate("to") .. " '" .. outputFile .. "'"`
Optimal result would look in german like this:
`$FilePathA konnte nicht nach $FilePathB verschoben werden.`

Maybe switch to translating a single string per message and then do placeholder replacement instead of concatenating single parts?

https://github.com/openwrt/luci/blob/master/applications/luci-app-simple-adblock/luasrc/model/cbi/simple-adblock.lua

Thank you!
Cheers, Chris

jow-, 2020-04-10:
@stangri - please change the source to translate format string:

`translatef("failed to move '%s' to '%s'", outputCache, outputFile)`

ce-4, 2020-04-10:
I created https://github.com/openwrt/luci/pull/3875
Please check if all is correct as i only did a syntax check

jow-, 2020-04-11:
I assume this has been addressed with the various i18n improvements.

closed 2020-04-11 by jow-